NicoHood/HID #132: Undefined reference to 'Keyboard´

Mauscorpio, 2018-04-06:
Hi all,

I'am havind this issue, and I have checked all the code and don't know where the error is.  Does anyone have had this issue?

The compilling error is:
sketch/SingleReport/../HID-APIs/KeyboardAPI.hpp:80: undefined reference to `Keyboard'
sketch/SingleReport/../HID-APIs/KeyboardAPI.hpp:80: undefined reference to `Keyboard'
sketch/SingleReport/../HID-APIs/KeyboardAPI.hpp:80: undefined reference to `Keyboard'
sketch/SingleReport/../HID-APIs/KeyboardAPI.hpp:80: undefined reference to `Keyboard'
/tmp/ccojDIbc.ltrans0.ltrans.o: In function `KeyboardAPI::press(KeyboardKeycode) [clone .constprop.17]':
sketch/SingleReport/../HID-APIs/KeyboardAPI.hpp:60: undefined reference to `Keyboard'
/tmp/ccojDIbc.ltrans0.ltrans.o:sketch/SingleReport/../HID-APIs/KeyboardAPI.hpp:60: more undefined references to `Keyboard' follow
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board HoodLoader2 16u2.

I'm Using an arduino Uno, and it is already flashed to Hoodloader2

Help please.

NicoHood, 2018-04-07:
Please post the full error report. please enabke versbose mode

Mafields, 2018-06-02:
I have nearly same error...

Arduino: 1.8.5 (Windows 10), Board: "Arduino Leonardo"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\xxxx\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\xxxx\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\xxxx\Documents\Arduino\libraries -fqbn=arduino:avr:leonardo -ide-version=10805 -build-path C:\Users\xxxx\AppData\Local\Temp\arduino_build_450588 -warnings=default -build-cache C:\Users\xxxx\AppData\Local\Temp\arduino_cache_526947 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avr-gcc.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -prefs=runtime.tools.avrdude.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -verbose C:\Users\xxxx\Desktop\volumeknob2\volumeknob2.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\xxxx\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\xxxx\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\xxxx\Documents\Arduino\libraries -fqbn=arduino:avr:leonardo -ide-version=10805 -build-path C:\Users\xxxx\AppData\Local\Temp\arduino_build_450588 -warnings=default -build-cache C:\Users\xxxx\AppData\Local\Temp\arduino_cache_526947 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avr-gcc.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -prefs=runtime.tools.avrdude.path=C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -verbose C:\Users\xxxx\Desktop\volumeknob2\volumeknob2.ino
Using board 'leonardo' from platform in folder: C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21
Using core 'arduino' from platform in folder: C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21
Detecting libraries used...
"C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-IC:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\leonardo" "C:\Users\xxxx\AppData\Local\Temp\arduino_build_450588\sketch\volumeknob2.ino.cpp" -o "nul"
"C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-IC:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\leonardo" "-IC:\Users\xxxx\Documents\Arduino\libraries\HID" "C:\Users\xxxx\AppData\Local\Temp\arduino_build_450588\sketch\volumeknob2.ino.cpp" -o "nul"
"C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-IC:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\leonardo" "-IC:\Users\xxxx\Documents\Arduino\libraries\HID" "-IC:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\HID\src" "C:\Users\xxxx\AppData\Local\Temp\arduino_build_450588\sketch\volumeknob2.ino.cpp" -o "nul"
Using cached library dependencies for file: C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\HID\src\HID.cpp
Generating function prototypes...
"C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0 [... cut ...]
C:\Users\xxxx\AppData\Local\Temp\cc21ph3h.ltrans0.ltrans.o:C:\Users\xxxx\Documents\Arduino\libraries\HID/SingleReport/../HID-APIs/KeyboardAPI.hpp:60: more undefined references to `Keyboard' follow

collect2.exe: error: ld returned 1 exit status

Using library HID in folder: C:\Users\xxxx\Documents\Arduino\libraries\HID (legacy)
Using library HID at version 1.0 in folder: C:\Users\xxxx\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\HID 
exit status 1
Error compiling for board Arduino Leonardo.