NicoHood/HID #133: Expected to work?

MartinYoungOB, 2018-04-16:
I've been trying to get raw HID. It's partially working but not quite. I'm wondering if it's expecting to work and if not, what I need to do to make it work? I've been looking through the source and there are quite a comments that suggest that it might not be complete. I don't understand many of them.

I have a genuine Arduino Uno with a 16u2. I've put Hoodloader2 on it and it works exactly as expected - I can upload programs to it and communicate with the other AVR on the same board.

I'm now trying to use RawHID to set up a bidirectional link between my Uno and the PC using my own vendor id and product id. I'm using hidapi on Windows 10 (from the main repo) on the PC.

When I enumerate devices on the PC, I see the 16u2 and I can connect to it. I can send bytes with hid_write() on the PC and receive them using RawHID.read() on the 16u2. This is great. (hidapi sends on the first OUT endpoint or the control endpoint is there isn't an OUT one. I'm not sure which it happening this case.)

However when I try to send from the 16u2 using RawHID.write() it is not received by the PC. Using Wireshark I can see that each character is arriving as the single payload byte in a URB_INTERRUPT packet but hid_read() does not return them.

Is this expected to work? Is there something else I need to do?

Also:

 - RawHID doesn't work in a multi-report interface. Because of the serial port that the 16u2 is implemented is this a multi-report system. Do I need to turn off the serial port somehow?

 - Can I induce RawHID to send on endpoint 0, the control endpoint?

Thanks,
Martin.

NicoHood, 2018-04-16:
RawHID support was always unstable. Also there were bugs introduced with the newer IDE versions which should be corrected now. But I've never retested it and its been quite a long time since I wrote the code.

You can simply just try it yourself and find the bugs. I sadly have no time to do so.

Multireport does not work with hidapi. It just confuses the API. But I am not using multireports, so nothing to do here.

RawHID receives data on a dedicated endpoint and sends data on endpoint0. You could send it on its own endpoint, but this library is not implemented like that.

NicoHood, 2018-04-17:
Your wireshark observation is correct and expected.

eivan7538, 2018-10-25:
Nico, Martin, did you have any success with Raw HID? I just tried it with the basic rawhid_test.c with Arduino MKR Zero and had the same issue: Arduino can receive packets from PC but when Arduino sends packets the PC does not receive anything. I would love to help but I'm an amateur so I'm not sure if I can debug and fix it.

zaphodbe, 2019-09-23:
I seem to have a similar situation.  It appears that if I do a RawHID.write(buf,len) with len less than 64 nothing gets sent, no LED flash nothing.  If however I use 64 or larger it does send, but looks like only the first 64 bytes.

Looking at the code it appears to just be calling USB_Send so assuming this is a lower level USBCore issue.

IDE 1.8.10 with Arduino library 1.8.1 the default in the 1.8.10 package.

spuder, 2022-08-07:
For anyone else running into this problem. Check the following

1. Read this sparkfun tutorial about the Raw HID library

https://learn.sparkfun.com/tutorials/hid-control-of-a-web-page/all

Set/Verify the `RAWHID_TX_SIZE` and `RAWHID_RX_SIZE` in the `usb_private.h` file (On my machine it is located at `./packages/framework-arduinoteensy/cores/usb_rawhid/usb_private.h` and both values are set to `64`)

2. Try filling the array buffer with all zeros

```
uint8_t rawhidData[64];
void setup()
  for (int i = 0; i < sizeof(rawhidData); i++) {
    rawhidData[i] = 0;
  }
}
```

3. Compile and use this USB HID viewer as a debugging tool. https://www.pjrc.com/teensy/rawhid.html

mcuee, 2023-06-17:
It seems to work reasonaby well based on my testing using the example here.
* https://github.com/todbot/hidapitester/issues/20

There is a bug for the Feature report but the fix is simple.
* https://github.com/NicoHood/HID/issues/418