NicoHood/HID #134: Keyboard HID

abigliazzi, 2018-04-19:
Hi NicoHood, im Angelo from  Argentina 

i need your help with a very very  short code  😺

i have this code

#include "Keyboard.h"
#include "HID.h"
void setup() {
  Keyboard.begin();
}

void loop() {
delay (1000);
  Keyboard.write(KEY_F1);
delay (3000);
  Keyboard.write(KEY_F2);

}

this code works fine for me... but the problem is  

this code give me only one "F1" and one "F2"  in 4 seconds.

i need "F1" appears every second (1000ms) and "F2" every 3 seconds (3000ms)

in a infinyte loop

gedankenexperimenter, 2018-04-19:
```
byte counter = 0;
delay(1000);
Keyboard.write(KEY_F1);
if (counter == 2) {
  Keyboard.write(KEY_F2);
  counter == 0;
} else {
  ++counter;
}
```

abigliazzi, 2018-04-19:
Thank you, only work F1 every 1000 ms , F2 doesnt work

abigliazzi, 2018-04-19:
```
#include "Keyboard.h"
#include "HID.h"

void setup() {

  Keyboard.begin();
}

void loop() {

byte counter = 0;
delay(1000);
Keyboard.write(65);
if (counter == 2) {
  Keyboard.write(66);
  counter = 0;
} else {
  ++counter;
}
}
```

I tried with key A and B (65 and 66)... only work "A" every 1000ms

NicoHood, 2018-04-19:
This is not a support forum, please ask this in the arduino forum or irc channel. Thanks.
Checkout the blink without delay example from ardino.

closed 2018-04-19 by NicoHood