NicoHood/HID #149: Digispark error

jfue4, 2018-09-17:
Arduino: 1.8.6 (Windows 7), Board: "Digispark (Default - 16.5mhz)"

In file included from C:\Users\User\Documents\sketch_sep17b\sketch_sep17b.ino:1:0:

C:\Users\User\Documents\Arduino\libraries\HID-Project\src/HID-Project.h:35:2: error: #error HID Project can only be used with an USB MCU.

 #error HID Project can only be used with an USB MCU.

  ^

In file included from C:\Users\User\Documents\Arduino\libraries\HID-Project\src/HID-Project.h:39:0,

                 from C:\Users\User\Documents\sketch_sep17b\sketch_sep17b.ino:1:

C:\Users\User\Documents\Arduino\libraries\HID-Project\src/SingleReport/SingleAbsoluteMouse.h:28:17: fatal error: HID.h: No such file or directory

 #include "HID.h"

                 ^

compilation terminated.

exit status 1
Error compiling for board Digispark (Default - 16.5mhz).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

blahlicus, 2018-09-18:
The Digispark is not compatible with this library. Only the following devices are compatible:

* Uno (with HoodLoader2)
* Mega (with HoodLoader2)
* Leonardo
* (Pro)Micro
* Any other 8u2/16u2/at90usb8/162/32u2/32u4 compatible board
* Any SAMD21 boards

I checked the attiny's spec sheet, I don't think the hardware is even capable of enumerating an HID device.

NicoHood, 2018-09-18:
@blahlicus Please open a PR ;)

closed 2018-09-18 by NicoHood