NicoHood/HID #151: Consumer status leds?

pmck91, 2018-09-21:
Is it possible to add support for consumer leds such as mute indicator. 

`#define LED_SHIFT                          0x07`
`#define LED_MUTE                0x09 // OOC - Indicates that the amplifier audio is shut off, 11.3 Consumer Indicators`
`#define LED_MICROPHONE                  0x21`
`#define LED_PAUSE                0x37`
`#define LED_PLAY                0x36`
`#define LED_STOP                0x33`
`#define LED_FAST_FORWARD           0x35`
`#define LED_REWIND             0x34`
`#define LED_REPEAT                0x10`

Ive seen these defined elsewhere but they clash with whats in improvedKeyboard.h

would adding something like 

`	LED_DO_NOT_DISTURB	= (1 << 7),`
`       LED_MUTE            = 0x09,`

help?

chopsuwe, 2019-09-27:
I'm giving this a go but not having much luck. The problem I'm running into is the bit shifting. The KeyboardLeds are stored as an 8 bit unsigned int, allowing up to 7 LEDs. We need 77 bits to cover all the LEDs in the usage table but I'm not sure how to increse the limit past 14. Declaring them as an 16 bit unsigned int seems to work (although is untestable since Windows doesn't appear to have native functionality for LED_MUTE 0x09 through to LED_REPEAT 0x10). Incresing to a 32 bit unsigned int causes a complier error. 

```
d:\files\arduino\libraries\hid-master\src\keyboardlayouts\improvedkeylayouts.h:70:22: error: enumerator value -32768 is outside the range of underlying type 'uint32_t {aka long unsigned int}'

 LED_REPEAT = (1 << 15),
```

```
enum KeyboardLeds : uint32_t {
  LED_NUM_LOCK		= (1 << 0),
  LED_CAPS_LOCK		= (1 << 1),
  LED_SCROLL_LOCK		= (1 << 2),
...
  LED_SURROUND_FIELD_ON	= (1 <<	14),
  LED_REPEAT	= (1 <<	15),    // This is line 70. Anything beyond here causes the compiler error. 
  LED_STEREO	= (1 <<	16),
  LED_SAMPLING_RATE_DETECT	= (1 <<	17),
};*/
```

I'm using a Pro Micro ATmega32U4 and I presume it can only handle 16 bit numbers. Delaring the LEDs as their hex values (LED_NUM_LOCK	=	0x01,) doesn't work at all. It compiles but the Arduino doesn't respond to the LEDs at all. 

How can the code be modified to allow the use of the hex values or bit shifting above 14 bits?

NicoHood, 2019-09-29:
Try `(uint32_t) (1 << 15)`

chopsuwe, 2019-10-02:
That's what gives the compiler error. I also tried declaring them using their hex values like some of the other libraries. 
```
enum KeyboardLeds : uint8_t {
LED_NUM_LOCK	=	0x01,
... };
```
That method doesn't work at all, it looks like other modifications would need to be made to the code but I don't know enough about programming to know what.

NicoHood, 2019-11-15:
did you try my code? The cast to uint32_t is important

chopsuwe, 2019-11-18:
Is this what you mean? I'm not sure I've done it right but it does compile without errors even when using LED_CAMERA_ON which is (1 << 39).  I'll test it next time I'm in the same location as the hardware. 

`enum KeyboardLeds : uint32_t {
  LED_NUM_LOCK    = (1 << 0),
...
  LED_SURROUND_FIELD_ON = (1 << 14),
  LED_REPEAT = (uint32_t)(1 << 15),    // From here on we use the cast. . 
  LED_STEREO = (uint32_t)(1 << 16),
...
  LED_SYSTEM_SUSPEND = (uint32_t)(1 << 75),
  LED_EXTERNAL_POWER_CONNECTED = (uint32_t)(1 << 76),
};`

NicoHood, 2020-01-27:
did it work?

chopsuwe, 2020-03-04:
I haven't been able to find any way to test it. All I can say is it doesn't break any functionality. 

Windows 8.1 & 10 only appear to support Caps, Num and Scroll lock. None of the other indicators I tried worked (speaker mute, play, battery, etc.). There may be support for the Compose and Kana LEDs in some versions but I haven't found a way to activate them. They don't seem to be in common use anyway, even on Japanese keyboards which makes me suspect support has either been removed or only applies to Japanese versions of the OS.

The inspiration for this project was to use the LEDs to provide a ring indication for the softphone at work. However this has been confirmed not to support the hook flash button on headsets so it's no surprise that it doesn't appear to send and LED signals either. 

I'd be interested to hear from anyone who can do testing and provide feedback.

yahesh, 2020-05-02:
Do you really think that something like `(uint32_t)(1 << 75)` makes sense at all? I mean, `uint32_t` has a size of 32 bits but you're trying to shift to bit 76 here. I guess, this will just lead to a value of `0`.

Jens1985, 2021-05-16:
Any news here?

This works with my Jabra headset. There an LED flashes for incoming calls (MS Teams, Jabber).
https://developer.jabra.com/site/global/sdks/web/documentation/index.gsp