NicoHood/HID #152: NKRO and Improved Keyboard not working on samd21

thnikk, 2018-09-27:
Boot keyboard works fine, but NKRO keyboard causes the keys to do nothing and Improved Keyboard locks up the board and gives a USB error on Windows. I'm testing all of them the same way by changing boot to nkro or removing it. My code is here and I'm using Adafruit's ItsyBitsy M0: https://github.com/thnikk/touchPad/blob/master/touchPad/touchPad.ino

Thanks for adding samd21 support in the first place!

NicoHood, 2018-09-29:
To debug it would be better to have a smaller test case. Can you please try if the examples work for you?

@facchinm Said the fix was tested properly, maybe he can help out?

thnikk, 2018-09-29:
Sorry about that, I was actually thinking the same thing. Here's some simplified code that has the same result, where boot keyboard works fine but nkro does nothing. Improved keyboard doesn't seem to be locking up the board anymore, so that was probably just an issue on my end. I tested this on both the Adafruit Trinket M0 and the ItsyBitsy M0 for good measure.

https://gist.github.com/thnikk/c7d425622b64fdb3562c76710529a7f9

NicoHood, 2018-09-30:
Did you test the nkro keyboard with an avr? I heard, that its broken by recent windows versions or similar. In general i would not recommend using it. It was just a proof, that its possible to build such a keyboard, but the usecases are really rare. Normally it is hard for the OS to handle more than 6 keypresses anyways.

thnikk, 2018-10-01:
It works exactly as expected on AVR. I just tested it with this code and it worked perfectly (using a 32u4 board): https://gist.github.com/thnikk/4e336c8dbed5c5ef3d875da19c046c49

I'm using the library for a game controller that needs up to 7 keys to be able to be pressed simultaneously.

NicoHood, 2018-10-02:
I cannot help, please try to debug it yourself. I am sorry. I do not own any of those boards, nor do I have time to fix it. I hope the community will help. But as I said, the normal keyboard should be good enough for most use cases.