NicoHood/HID #156: Consumer commands ignored on Mac OS

dherczeg-tgm, 2018-12-05:
The title says it all. Consumer commands seem to be ignored on Mac OS. They work fine on Windows 7 and 10 though. Mac OS version: 10.14 (Mojave), haven't tested it with other versions.

Arduino gets detected correctly and the code works fine (tested with serial outputs), but every consumer command (like: Consumer.press(MEDIA_VOL_DOWN);) gets ignored.

Board used: Pro Mini, tried changing the PID and VID to an apple keyboard too, same result.

WhiteBr0wnie-24, 2018-12-05:
> Did you try the single consumer version?
> It is possible, that this is a macos bug.

No I didn‘t. Do you have a link where I can download the single consumer version?
Thanks.

WhiteBr0wnie-24, 2018-12-05:
I tried it now and it didn't work either.

closed 2019-05-09 by WhiteBr0wnie-24