NicoHood/HID #166: question: How to judge whether the computer is turned on or off?

Antecer, 2019-03-06:
I will keep the keyboard connected every day.
I added some cool lights to it.
But, the backlight is still on after turning off the computer.
So, when the computer is turned off, does it send a specific command to the USB device?
Then i can use MCU get the command to turn off led light.

This problem is only for HID devices, i don't want to use any computer software to implement it.

underwoodblog, 2019-03-06:
Look at the BIOS of the computer. In some you can set, if the usb-ports are powered, when the computer is off. Maby try a other usb port on your computer, not all are powered in off mode or are you using a powered usb hub?

Antecer, 2019-03-06:
@underwoodblog
Sorry, I mean, in the case of not using the BIOS to turn off the USB power.
Does HID have any commands in this regard?

Antecer, 2019-03-06:
My purpose is not to disconnect the power to the device. I want
Use the MCU to know if the computer is turned off so that I can do some special work.

NicoHood, 2019-03-09:
You could try this:
https://github.com/arduino/ArduinoCore-avr/blob/6ba4fd363f8dcd704b73d6b4dd897fce6e7bfa72/cores/arduino/USBCore.cpp#L794

NicoHood, 2019-03-09:
Or if you are using lufa:
http://fourwalledcubicle.com/files/LUFA/Doc/140302/html/group___group___device.html#ga86f959954d90ef0227bc217d1473ccfa

Antecer, 2019-03-09:
@NicoHood Did you mean check USBConnected?
I think this can only determine whether the device has a valid address assigned by the host, but when the host is shut down, the address obtained by the device will not be reset.

NicoHood, 2019-03-09:
The Lufa core does all that for you. It is more stable and I think you can detect the connected host much better than with the arduino environment. Just try it. That are the options you have.

Closing as this is no issue with hid.

closed 2019-03-09 by NicoHood