NicoHood/HID #167: what is the position data in the surface dial report for?

OmegaRogue, 2019-03-10:


NicoHood, 2019-03-14:
@hansmbakker could you please answer? He wrote that code.

hansmbakker, 2019-03-14:
You can put the official surface device on the surface studio screen and then windows will draw the surface dial menu around the device.

In the protocol it has this meaning:
- if you set it, it lets windows know where on the screen you placed your dial device
- if it is not set, Windows thinks the dial device is not placed on the screen but e.g. on the desk.

Please refer to the [official docs](https://docs.microsoft.com/en-us/windows-hardware/design/component-guidelines/radial-controller-protocol-implementation) for more info. Also note that I did *not* implement the position part of the protocol (it is optional), but only the button and rotation properties (which are required).

hansmbakker, 2019-03-14:
Feel free to submit a pull request if you implement it, though :)

OmegaRogue, 2019-03-17:
but how does the surface dial know where it is placed?

OmegaRogue, 2019-03-17:
and im implementing it

OmegaRogue, 2019-03-17:
> and im implementing it

https://github.com/NicoHood/HID/pull/169

hansmbakker, 2019-03-17:
> but how does the surface dial know where it is placed?
I have no idea. [This pdf](https://news.microsoft.com/windows-event-2016-assets/Surface-Dial-Fact-Sheet.pdf) says that it is done by PixelSense technology (which is just the name of the surface screens, so it gives no explanation).

https://docs.microsoft.com/en-us/windows-hardware/design/component-guidelines/radial-controller-designs says that it is vendor-specific and also gives no details.

[This video](https://youtu.be/JzqLCIwIJYQ?t=71) at 1:11 shows a digitizer plate. I suspect there is an (active) matrix in the Surface screens that the Dial can detect, just like a stylus could work?

hansmbakker, 2019-03-17:
By the way, what are you planning to use the position properties, and the Dial code in general, for?

OmegaRogue, 2019-03-18:
> By the way, what are you planning to use the position properties, and the Dial code in general, for?

A diy surface dial like hid device

hansmbakker, 2019-03-18:
Nice, good luck :)

OmegaRogue, 2019-03-18:
what are the indentation guidelines for the code? tabs or 4 spaces?

hansmbakker, 2019-03-18:
This is a question for @nicohood, I just meant to ask to make it consistent.

OmegaRogue, 2019-03-18:
because in some files its 2 spaces, in some its tabs

NicoHood, 2019-03-30:
It should be spaces for every software project. Feel free to open a PR that fixes all tabs. This is possibly left over from very old files. 2 or 4 spaces.. depends on the other code. For new projects I am using 4 spaces nowadays.

OmegaRogue, 2019-03-30:
i always use tabs