NicoHood/HID #172: platform.h no such file or directory

aganim4, 2019-05-13:
Hello there!

Unfortunately, I am not super experienced with programming. I found this one thread with a similar problem but it did not work for me, or I did not know how to implement this for my case.

I am using an arduino uno r3 clone (himalaya uno rev3) and installed the bootloader correctly (or at least I think so. It is displayed in devices and printers and has number 0205.
Now, when I try to load the keyboard example I get:

> C:\Program Files (x86)\Arduino\libraries\HID\Keyboard.cpp:26:22: fatal error: Platform.h: No such file or directory

compilation terminated.

exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Uno.

I put the HID-master without "-master" into C:\Program Files (x86)\Arduino\libraries and have no idea what to do now... I just want to use the arduino to emulate a keyboard.

sorry if this is probably a super basic problem but I have no idea where to look for...

thanks!!

NicoHood, 2019-05-28:
Please install via the library manager.

NicoHood, 2019-05-28:
https://github.com/NicoHood/HID/wiki/Installation

closed 2019-05-28 by NicoHood