NicoHood/HID #174: AbsoluteMouse.MoveTo not moving to position

cyberfizz, 2019-05-31:
I apologise if i am doing something wrong but i am trying to move the mouse based on ethernet commands. It looks like the mouse moves but doesn't move to the right positions.

Am i doing something wrong??

`String getValue(String data, char separator, int index)
{
    int found = 0;
    int strIndex[] = { 0, -1 };
    int maxIndex = data.length() - 1;

    for (int i = 0; i <= maxIndex && found <= index; i++) {
        if (data.charAt(i) == separator || i == maxIndex) {
            found++;
            strIndex[0] = strIndex[1] + 1;
            strIndex[1] = (i == maxIndex) ? i+1 : i;
        }
    }
    return found > index ? data.substring(strIndex[0], strIndex[1]) : "";
}
void processCommand(String com, EthernetClient c)
{
  com.trim();
  //c.println(com);
  if(com.startsWith("movemouse")){
      String xval = getValue(com,';',1);
      String yval = getValue(com,';',2);
      
      int x = xval.toInt();
      int y = yval.toInt();

      SingleAbsoluteMouse.moveTo(x, y);
      c.println(String("AbsoluteMouse.moveTo(" + String(x) + ", " + String(y) + ");"));

      c.println("OK");     
  }
}`

I have tried normal AbsoluteMouse and Single too but both the same. I also tried basing it on long int size but seems to always stick in the top corner of display. 0,0 seems to work though.

The command being sent through is movemouse;X;Y and then the string gets made in to int for the commands. The commands look ok too. Am i missing some scaling or something?

NicoHood, 2019-06-04:
This is not a support forum, this question belongs to the arduino forums or chat. please ask there, I dont have time teaching you programming. If there is a problem with the library, let me know.

closed 2019-06-04 by NicoHood

aidv, 2024-05-17:
> This is not a support forum, this question belongs to the arduino forums or chat. please ask there, I dont have time teaching you programming. If there is a problem with the library, let me know.

OP is actually right.

`moveTo()` is not moving the cursor to the right position.

I am experiencing the same issue as OP.

Your attitude is disgusting. GFYS

aidv, 2024-05-17:
> I apologise if i am doing something wrong but i am trying to move the mouse based on ethernet commands. It looks like the mouse moves but doesn't move to the right positions.
> 
> Am i doing something wrong??
> 
> ```c
> String getValue(String data, char separator, int index)
> {
>     int found = 0;
>     int strIndex[] = { 0, -1 };
>     int maxIndex = data.length() - 1;
> 
>     for (int i = 0; i <= maxIndex && found <= index; i++) {
>         if (data.charAt(i) == separator || i == maxIndex) {
>             found++;
>             strIndex[0] = strIndex[1] + 1;
>             strIndex[1] = (i == maxIndex) ? i+1 : i;
>         }
>     }
>     return found > index ? data.substring(strIndex[0], strIndex[1]) : "";
> }
> void processCommand(String com, EthernetClient c)
> {
>   com.trim();
>   //c.println(com);
>   if(com.startsWith("movemouse")){
>       String xval = getValue(com,';',1);
>       String yval = getValue(com,';',2);
>       
>       int x = xval.toInt();
>       int y = yval.toInt();
> 
>       SingleAbsoluteMouse.moveTo(x, y);
>       c.println(String("AbsoluteMouse.moveTo(" + String(x) + ", " + String(y) + ");"));
> 
>       c.println("OK");     
>   }
> }
> ```
> 
> I have tried normal AbsoluteMouse and Single too but both the same. I also tried basing it on long int size but seems to always stick in the top corner of display. 0,0 seems to work though.
> 
> The command being sent through is movemouse;X;Y and then the string gets made in to int for the commands. The commands look ok too. Am i missing some scaling or something?

Did you manage to resolve this? I'm having the same issue right now.

cyberfizz, 2024-05-17:
> Did you manage to resolve this? I'm having the same issue right now.

I didnt unfortunately and scrapped the program. I have done a lot of work with KVM though in the last year or so and suspect its because windows needs a special driver or something to enable absolute mouse mode. http://www.blackbox.com/en-ca/insights/blogs/detail/technology/2021/01/27/absolute-mouse-mode-explained#:~:text=Microsoft%20Windows%2010%20added%20ABS,ABS%20Mode%20to%20function%20properly.

aidv, 2024-05-17:
> I didnt unfortunately and scrapped the program. I have done a lot of work with KVM though in the last year or so and suspect its because windows needs a special driver or something to enable [...]

Interesting. I'm working on a KVM project right now.

Want to connect and potentially join?

cyberfizz, 2024-05-17:
> Interesting. I'm working on a KVM project right now.
>
> Want to connect and potentially join?

Im not developing something sorry, working with off the shelf KVM options. Thank you anyway for the offer, good luck with your project