NicoHood/HID #175: On switching language specific input method

blahlicus, 2019-06-25:
@wladimir-computin I see you are the author behind the generic input method support, would you mind providing some insight on how to use this feature?

Suppose I want to use the DE layout which you have provided, do I `#define LAYOUT_GERMAN` in my Arduino sketch or will I have to edit HID-Settings.h in the library itself?

Also, I see in your HID-Settings.h that you have the following:

```
//================================================================================
// Settings
//================================================================================

#define LAYOUT_US_ENGLISH
//#define LAYOUT_CANADIAN_FRENCH
//#define LAYOUT_CANADIAN_MULTILINGUAL
//#define LAYOUT_DANISH
//#define LAYOUT_FINNISH
//#define LAYOUT_FRENCH
//#define LAYOUT_FRENCH_BELGIAN
//#define LAYOUT_FRENCH_SWISS
//define LAYOUT_GERMAN
//#define LAYOUT_GERMAN_MAC
//#define LAYOUT_GERMAN_SWISS
//#define LAYOUT_ICELANDIC
//#define LAYOUT_IRISH
//#define LAYOUT_ITALIAN
//#define LAYOUT_NORWEGIAN
//#define LAYOUT_PORTUGUESE
//#define LAYOUT_PORTUGUESE_BRAZILIAN
//#define LAYOUT_SPANISH
//#define LAYOUT_SPANISH_LATIN_AMERICA
//#define LAYOUT_SWEDISH
//#define LAYOUT_TURKISH
//#define LAYOUT_UNITED_KINGDOM
//#define LAYOUT_US_INTERNATIONAL

//================================================================================
// Definitions and Helpers
//================================================================================

// Default US keyboard layout
#if !defined(LAYOUT_CANADIAN_FRENCH) && !defined(LAYOUT_CANADIAN_MULTILINGUAL) \
&& !defined(LAYOUT_DANISH) && !defined(LAYOUT_FINNISH) && !defined(LAYOUT_FRENCH) \
&& !defined(LAYOUT_FRENCH_BELGIAN) && !defined(LAYOUT_FRENCH_SWISS) && !defined(LAYOUT_GERMAN) \
&& !defined(LAYOUT_GERMAN_MAC) && !defined(LAYOUT_GERMAN_SWISS) && !defined(LAYOUT_ICELANDIC) \
&& !defined(LAYOUT_IRISH) && !defined(LAYOUT_ITALIAN) && !defined(LAYOUT_NORWEGIAN) \
&& !defined(LAYOUT_PORTUGUESE) && !defined(LAYOUT_PORTUGUESE_BRAZILIAN) \
&& !defined(LAYOUT_SPANISH) && !defined(LAYOUT_SPANISH_LATIN_AMERICA) \
&& !defined(LAYOUT_SWEDISH) && !defined(LAYOUT_TURKISH) && !defined(LAYOUT_UNITED_KINGDOM) \
&& !defined(LAYOUT_US_INTERNATIONAL) && !defined(LAYOUT_US_ENGLISH)
#define LAYOUT_US_ENGLISH
#endif
```

Why is it necessary to have the first `#define LAYOUT_US_ENGLISH`? Shouldn't the lack of any defined layout cause it to default to the ASCII layout as per the last line of code?

wladimir-computin, 2019-06-27:
Hey  @blahlicus ,

yes, the feature is written by me. Here's how to add and use another keyboard layout:

1. Copy one of the ImprovedKeylayoutsDE.h or ImprovedKeylayoutsUS.h and rename it to ImprovedKeylayoutXX.h, where XX is your locale.

2. Edit the ImprovedKeylayoutXX.h file to match your scancodes. Try'n'error most of the time.
You can add new non-ASCII characters like I did for german Umlaute. But keep in mind, that if you're Arduino sketch uses them, you can't switch on the fly between the layouts anymore since other layouts will have missing defines for your sketch.

3. In ImprovedKeylayout.h, edit this part to include your layout if you define it.
```
#if defined(LAYOUT_US_ENGLISH)
	#include "ImprovedKeylayoutsUS.h"
#elif defined(LAYOUT_CANADIAN_FRENCH)
#elif defined(LAYOUT_CANADIAN_MULTILINGUAL)
#elif defined(LAYOUT_DANISH)
#elif defined(LAYOUT_FINNISH)
#elif defined(LAYOUT_FRENCH)
#elif defined(LAYOUT_FRENCH_BELGIAN)
#elif defined(LAYOUT_FRENCH_SWISS)
#elif defined(LAYOUT_GERMAN)
	#include "ImprovedKeylayoutsDE.h"
#elif defined(LAYOUT_GERMAN_MAC)
#elif defined(LAYOUT_GERMAN_SWISS)
#elif defined(LAYOUT_ICELANDIC)
#elif defined(LAYOUT_IRISH)
#elif defined(LAYOUT_ITALIAN)
#elif defined(LAYOUT_NORWEGIAN)
#elif defined(LAYOUT_PORTUGUESE)
#elif defined(LAYOUT_PORTUGUESE_BRAZILIAN)
#elif defined(LAYOUT_SPANISH)
#elif defined(LAYOUT_SPANISH_LATIN_AMERICA)
#elif defined(LAYOUT_SWEDISH)
#elif defined(LAYOUT_TURKISH)
#elif defined(LAYOUT_UNITED_KINGDOM)
#elif defined(LAYOUT_US_INTERNATIONAL)
#endif
```

4. In HID-Settings.h, define your layout by uncommenting the matching line to use your layout:

```
//================================================================================
// Settings
//================================================================================

#define LAYOUT_US_ENGLISH
//#define LAYOUT_CANADIAN_FRENCH
//#define LAYOUT_CANADIAN_MULTILINGUAL
//#define LAYOUT_DANISH
//#define LAYOUT_FINNISH
//#define LAYOUT_FRENCH
//#define LAYOUT_FRENCH_BELGIAN
//#define LAYOUT_FRENCH_SWISS
//define LAYOUT_GERMAN
//#define LAYOUT_GERMAN_MAC
//#define LAYOUT_GERMAN_SWISS
//#define LAYOUT_ICELANDIC
//#define LAYOUT_IRISH
//#define LAYOUT_ITALIAN
//#define LAYOUT_NORWEGIAN
//#define LAYOUT_PORTUGUESE
//#define LAYOUT_PORTUGUESE_BRAZILIAN
//#define LAYOUT_SPANISH
//#define LAYOUT_SPANISH_LATIN_AMERICA
//#define LAYOUT_SWEDISH
//#define LAYOUT_TURKISH
//#define LAYOUT_UNITED_KINGDOM
//#define LAYOUT_US_INTERNATIONAL
```

wladimir-computin, 2019-06-27:
Unfortunately, you have to place the define in HID-Settings.h, because of the way Arduino compiles the stuff. Haven't found another solution yet.

wladimir-computin, 2019-06-27:
And yes, either the first define or the big if else block is unnecessary. It's like historically grown code, you know? ;)
But it prevents from failing to compile if you forget to define a layout.

alby696, 2019-11-06:
Hi everyone, 
I tried to copy the German keyboard and put it as an Italian keyboard also changing the main implementation of the keyboard "#include" ImprovedKeylayoutsIT.h "but it doesn't see it in any way and obviously it writes me what it wants as if it took always the US one.
Can anyone help me out, and above all say a program or a guide for a scancode key, so that I can make the correct file for the ITA keyboard?
Thanks