NicoHood/HID #177: Lock keys (Caps, Scroll, Num) are bugged in SAMD21 when using BootKeyboard if you press and hold them or press them repeatedly

blahlicus, 2019-06-28:
This is an SAMD21 and BootKeyboard specific issue.

Lock keys (caps, scroll, num) break the library and cause the entire library to bug out and spasm if you press and hold a lock key or press a lock key repeatedly.

If you press and hold a lock key, then release the lock key, your lock key will not be released, furthermore it disables keyboard inputs for other characters so that you are unable to send any other keystrokes forever until you reboot your device. Sometimes it does not disable all the keyboard inputs but it outputs weird characters as if the HIDreport was jumbled.

To reproduce this issue, use an SAMD21 board and run the following sketch:

```

#include "HID-Project.h"

void setup() {
  // put your setup code here, to run once:
  BootKeyboard.begin();
  delay(10000);
  BootKeyboard.press(KEY_SCROLL_LOCK);
  delay(2000);
  BootKeyboard.press(KEY_A);
  delay(2000);
  BootKeyboard.release(KEY_SCROLL_LOCK);
  delay(2000);
  BootKeyboard.release(KEY_A);
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Expected behaviour:

* Scroll lock is pressed
* a is pressed
* Scroll lock is released
* a is released

Current behaviour:

* Scroll lock is pressed
* a is pressed
* Scroll lock never gets released
* other characters cannot be sent via BootKeyboard.write or press
```

This is especially apparent if you have a few physical buttons hooked up to your SAMD21, if you set one of them to caps, then press it repeatedly or hold it down for a while, then it will completely disable your keypad until you unplug and replug your device.

NicoHood, 2019-07-05:
the underlying API should be the same. So the report should be fine, but is it really? Did you check that the sent keyboard report got cleared correctly?

If yes, then its a samd bug, maybe this can help: https://github.com/NicoHood/HID/issues/102

blahlicus, 2019-07-09:
It could be a hardware issue with my SAMD21 device because it is a custom PCB so maybe something is out of spec compared to the vanilla SAMD21 arduino.

I'll test this on the Arduino Zero and report back over the weekend.

NicoHood, 2019-07-30:
Any progress?