NicoHood/HID #183: question: how to use setFeatureReport

mindforger, 2019-08-30:
I try to wrap my head around the setFeatureReport part in the bootkeyboard with Hyperion rgb example.

How does this work? 

So if I want to make a small keyboard with some rgb leds how do I define the output and how do I feed data from the pc to the keyboard to change the colors? 

A hint to where I can find the other (pc) half of the example would be helping to understand I think.

NicoHood, 2019-09-23:
Hm what should I answer...
This feature is very experimental. It worked, yes. But I do not recommend to use it. If you really want to send additional data either write your own device with feature report support, use rawhid (which does not seem to work properly) or Serial. I'd use serial in your case.

The PC side example... Is a good question. I really dont know, my fault. :/ Sorry for no better answer.

mindforger, 2019-09-23:
continued in https://github.com/NicoHood/HID/issues/185 ... forgot i opened this one before :P

closed 2019-09-23 by mindforger