NicoHood/HID #185: Feature Report seems broken?!

mindforger, 2019-09-19:
I tried your Bootkeyboard Example on a ProMicro (Mega32U4, Arduino 1.8.9) and set up a Feature Report with 2 bytes total length including the ID byte and sent varoius length feature reports from Windows 10 using libusb/hidapi but it always returns -1 on the send_report and nothing is received on the device. Since i have no other device with known feature reports i can not tell if either the libusb/hidapi is doing somethign wrong, windows 10 doing funky stuff or the device failing, but i can use every other function of the hidapi without any issues.

Is there a way to get a debug printout of the configured descriptors to make sure everything is fine with them?!

NicoHood, 2019-09-23:
I guess you have to debug this on your own or simply use the serial port to send data.

mindforger, 2019-09-23:
> I guess you have to debug this on your own or simply use the serial port to send data.

yeah i was hoping to get an advice how to get access to the descriptors since i had issues to use include to access them without breaking the compiler, there seems to be some sequential include order that i have to deal with (sorry for not being clear on what i needed)

to the answer from the other issue (i forgot i opened that one before :P )

> use rawhid (which does not seem to work properly) or Serial. I'd use serial in your case.

i am tempted to follow the rabbit hole since i had a deeper look into the usb HID spec and it does not look THAT complicated but i guess "side effects" are what makes the rabbit hole here, especially with Win10

My main issues would be, that i need 2 serial ports, one for streamed data (or say binary data, feedback, displaying stuff, interrupt channel, ... ) and one like a console based configuration channel (i planned using some sort of console terminal style for basic configuration first and slap on an task tool later)

and getting 2 serials with arduino seems "impossible" to say the least (spent my whole weekend with trying librarys)

NicoHood, 2019-09-23:
If you try to use USB on a more advanced level i recommend using LUFA. With arduino you will not get satisfying results.

mindforger, 2019-09-23:
sounds promising, will give it a shot instead :)

to be honest working with VStudio directly makes more fun anyways than the restrictive arduino IDE, where you have to follow rabbit holes to find the interface specification to a lib or how to use it properly.

Sorry for fellow readers hoping to find some help on this like i did :(

closed 2019-09-23 by mindforger

NicoHood, 2019-09-23:
Well I do not use vidual studio at all, just the raw files. But that could work too.