NicoHood/HID #186: How to get game effect

hoantv, 2019-09-20:
I add more PDI descriptor to hdi discriptor game pad and using new pdi descriptor.
The gamepad work well. But how to send to get effect using the HDI library?

NicoHood, 2019-09-23:
I dont know what you are using/adding. I guess you have to do the magic and write some additional code on your own. What is the PDI descriptor used for?

hoantv, 2019-09-25:
I use PID descriptor from pid1_01.pdf. using your library (add ffb effect to gamepadAPI). Pc recognized it as ffb device. I will investigate add receiveReport.
In sendpo

hoantv, 2019-09-25:
I used PID descriptor in pid1_01.pdf. PC recongized it as ffb device.
Now I try add receiveReport(id,data,len).
In sendReport(id, data, len) id is Hidreportgamepad, so maybe id in receiveReport is effect report.

NicoHood, 2019-09-29:
Nice! Would you mind opening a PR with the FB support? I guess a lot of people would profit by this.

hoantv, 2019-09-29:
Atm i havent’t yet receive effect report. I saw many funcion in HId::settup but I don’t know which HID:: setup mean (dont see where it is called).
I did like this:
 - Add Rx_enpont = plugableusb+1
 - add interface for rx
 - add interupt out for rx.
- add function availabe check rx>0
- add get data for rx
- main program:
  + check avalable rx
  + read first byte check request
  + belong to request i receive, i send reponse to pc.

NicoHood, 2019-09-30:
Sure, but it would be easier to see your changes if you open a PR or at least share the modified files via file upload with us.

hoantv, 2019-09-30:
Sorry, i dont know PR mean. 
Here is my changes.
**In HID.h** add
#define HID_REPORT_TYPE_FEATURE 3
#define HID_ENPOINT_COUNT 2   //add
#define HID_ENDPOINT_IN   (pluggedEndpoint) //add
#define HID_ENDPOINT_OUT  (pluggedEndpoint+1) //add

  EndpointDescriptor  in;
  EndpointDescriptor  out; //add

  void AppendDescriptor(HIDSubDescriptor* node);
  int RecvReport(void* data, int len);; //add
  uint8_t AvailableReport();; //add
  
int SendControl(uint8_t flags, const void* d, int len);; //add
int RecvControl(void* d, int len);; //add
int RecvControlLong(void* d, int len);; //add

  uint8_t epType[HID_ENPOINT_COUNT]; //modify from 1 to 2
**In HID.ccp**

D_ENDPOINT(USB_ENDPOINT_IN(HID_ENDPOINT_IN), USB_ENDPOINT_TYPE_INTERRUPT, USB_EP_SIZE, 0x01),
    D_ENDPOINT(USB_ENDPOINT_OUT(HID_ENDPOINT_OUT), USB_ENDPOINT_TYPE_INTERRUPT, USB_EP_SIZE, 0x01) //add
 // add
int HID_::RecvReport(void* data, int len) {
  return USB_Recv(HID_ENDPOINT_OUT, &data, len);
}
 // add
uint8_t HID_::AvailableReport()  {
  return USB_Available(HID_ENDPOINT_OUT);
}

int HID_::SendControl(uint8_t flags, const void* d, int len){
  //not yet finish
}
int HID_::RecvControl(void* d, int len){
   //not yet finish
}
int HID_::RecvControlLong(void* d, int len){
 //not yet finish
}

[FfbWheel.zip](https://github.com/NicoHood/HID/files/3672415/FfbWheel.zip)

Yesterday, i make it receive data,  After change some thing i can't receive anymore. Many change so i can not rollback code :(! Now when check Wheel.availableReport, it alway =0.

hoantv, 2019-10-03:
@NicoHood Here is my code:
https://github.com/hoantv/VNWheel

EtienneGameSeed, 2019-10-04:
hoantv, maybe you don't know, but you are using code I've published a few years ago without my permission and without even explaining where it comes from. 
This is not correct !
Where did you get it ?

hoantv, 2019-10-04:
@NicoHood Sorry about that, Here are some reference code I got.
About WHID.h/WHID.cpp i modify from library HID after reading tutorial on microchip, some posts on adruino forums, some small codes i found on google (without comments where they came)
About wheel.h/wheel.cpp i use some codes from your gamepad (i will comment later or 
About ffb i use Etienne Saint-Paul‘s source code (already had comment).
It is not for commercial purpose. Sorry about that again.

EtienneGameSeed, 2019-10-06:
Could you please give me the link on google code ?

I've seen you cleaned up the unused code. Thanks. I don't know what you call the ffb engine, as there's no processing of the effects (The most important part of the code comes from adapt-ffb-joy project). You can keep the HID descriptor code, it's not rocket science...

hoantv, 2019-10-21:
@EtienneGameSeed
Here it is

https://drive.google.com/open?id=1CrJxms5pFNL6whzVFgNIWS55XVVezBFZ

EtienneGameSeed, 2019-10-21:
It's not a google code link...
It was uploaded 1 year ago by Fernando Ígor, is that you ?

EtienneGameSeed, 2019-10-21:
The problem is that you're not respecting original creators and licences. Open source doesn't work like that, you can't just take code that includes licences in the headers, remove them and publish it on github, which is what you did.
You removed references in the headers, as well as the non commercial creative commons licence in the main folder I published.