NicoHood/HID #189: Lets add support for 3D mouse!

Matesaktesak, 2019-09-23:
I would like if you guys, could help me with adding support for the 3D/Space mouse! I have already started, but progress is going slow, as I don't really understand what am I doing...

ubergeekseven, 2020-09-30:
I know this is old but, if you have a 3d mouse then you should understand the undertaking here. there is only 1 real 3d mouse available and they have custom drivers and hooks into software that the company and the software devs have to work together to have it function. The one case I have seen that is not this way is blender. It being open allows it though and the support is minimal and feels odd. You can approximate the actions you want with the camera and that is about it.
That being said, if you were to get all of the functions out of the 3dconnexion hid reports and add them to the library, it might function with the 3dconnexion driver itself. I do not see why it would not. I still want a 3d mouse in zbrush and there has been a fight in all the forums for years about how stupid it is to have support for it. I do not believe so but, the overwhelming voices hate the idea for some reason.

Anyway, I would love to be able to build my own space mouse. I can see lots of uses for it in different forms.

NicoHood, 2020-10-01:
You can test it here:
https://github.com/NicoHood/HID/pull/226

ubergeekseven, 2020-10-01:
@NicoHood I didn't know it was actually bring worked on. Thanks