NicoHood/HID #194: Cant get gamepad examples working on SAMD21

DCooper-nz, 2019-10-08:
I am trying to get the gamepad code working on a SAMD21 board and am having no joy. The code compiles and uploads fine, the board is identified in the windows usb controller setup but I get no data from the board. The same code works fine on a Leonardo. Also the keyboard and mouse examples both work fine on the SAMD21 just not the gamepad. I have tried multiple SAMD21 based boards and several computers with the same result.

thomasgauthier, 2019-10-12:
I have the same issue on the Adafruit Feather M0. Arduino Leonardo works fine.

jrowberg, 2019-11-10:
Same symptoms here. Could this be due to the relatively recent modifications to the USB implementation in the samd core (at least Adafruit’s fork, I think the official Arduino repo as well) which allows switching in the IDE to TinyUSB instead? I see the game pad with many axes and a zillion buttons, but nothing ever changes when advancing states via the button/interrupt.

EtienneGameSeed, 2019-11-11:
Same here, no refresh. Device appears as expected, but no report is received.

gdsports, 2020-01-19:
@DCooper-nz @thomasgauthier @jrowberg @EtienneGameSeed The gamepad works with this patch. The solution is the union/struct for the HID report must be packed to avoid the addition of extra filler bytes. Tested with Gamepad, Gamepad1, Gamepad2, Gamepad3, and Gamepad4 on Ubuntu 18.04 with jstest-gtk.

```
diff --git a/src/HID-APIs/GamepadAPI.h b/src/HID-APIs/GamepadAPI.h
index 6a50018..0a9fe4a 100644
--- a/src/HID-APIs/GamepadAPI.h
+++ b/src/HID-APIs/GamepadAPI.h
@@ -38,15 +38,16 @@ THE SOFTWARE.
 #define GAMEPAD_DPAD_LEFT 7
 #define GAMEPAD_DPAD_UP_LEFT 8
 
+#define ATTRIBUTE_PACKED  __attribute__((packed, aligned(1)))
 
-typedef union {
+typedef union ATTRIBUTE_PACKED {
        // 32 Buttons, 6 Axis, 2 D-Pads
        uint8_t whole8[0];
        uint16_t whole16[0];
        uint32_t whole32[0];
        uint32_t buttons;
 
-       struct{
+       struct ATTRIBUTE_PACKED {
                uint8_t button1 : 1;
                uint8_t button2 : 1;
                uint8_t button3 : 1;

```

DylanDophemont, 2020-03-02:
@gdsports thank you! Your patch works for my Adafruit Itsy Bitsy M0.

EtienneGameSeed, 2020-03-06:
Thanks, it worked for me too, after changing the PID.
First the device couldn't start, I had a code 10 error on Windows 10.
It seems Windows is keeping some things in cache that I could not remove. I tried to suppress the device and drivers, but that could not solve the problem.
Finally I changed the PID and it worked.
If anyone knows how to clean those info that windows is keeping, It would be fantastic.

engeen-nl, 2020-04-08:
> @DCooper-nz @thomasgauthier @jrowberg @EtienneGameSeed The gamepad works with this patch. The solution is the union/struct for the HID report must be packed to avoid the addition of extra filler bytes. Tested with Gamepad, Gamepad1, Gamepad2, Gamepad3, and Gamepad4 on Ubuntu 18.04 with jstest-gtk.

Thank you, this also worked for me using Windows 8. And it still works on MacOS then as well, so I would say release a new version of the library with this patch :)

tozz88, 2020-08-02:
NOTE: This issue is also in the AbsoluteMouse. I came here looking for my problem, adapted this solution to AbsoluteMouse.h
and it works!

NicoHood, 2020-08-02:
@tozz88 thanks for the note. I've added a comment to the open PR https://github.com/NicoHood/HID/pull/211

closed 2020-08-09 by NicoHood