NicoHood/HID #200: Multiple modifier keys + M key issues

electricrelay, 2020-01-13:
I'm having issues creating code to type a hot key Ctrl + Alt + m. I'm using keyboard testing software and it is only detecting the Ctrl and Alt key being pressed. the M key is completely ignored. 
I have this so far
`Keyboard.press(KEY_LEFT_CTRL);`
`Keyboard.press(KEY_LEFT_ALT);`
`Keyboard.press('m');`
`delay(100);`
`Keyboard.releaseAll();`

If I remove `Keyboard.press(KEY_LEFT_CTRL);` or `Keyboard.press(KEY_LEFT_ALT);` then it will type Ctrl + m / Alt + m respectively

If I change the alpha key to `p` or `a` or `b` or `c` (I havn't tested the whole alphabet) then it will work. 

In summary, I can't get Ctrl + Alt + m to work but Ctrl + m, Alt + m, and Ctrl + Alt +p do...

dyfcom, 2020-10-15:
I got the problem with exact the same code, that the SHIFT-Key is holding down, but doesn't release it. 
I used the default keyboard.h before and this worked fine.

dyfcom, 2020-10-15:
> 
> 
> I got the problem with exact the same code, that the SHIFT-Key is holding down, but doesn't release it.
> I used the default keyboard.h before and this worked fine.

Ok found the problem. It was, that I used two char's for a shorter variable for the keys, like in the Arduino example. After I deleted the char's and used the normal expression, it worked fine.

NicoHood, 2020-11-22:
@electricrelay could you solve your issue?

underwoodblog, 2020-12-03:
I have a similar issue with Bootkeyboard and CTRL+S, when I use it on a SoundDevice MixPre. All other Keys that I send, like CTRL+R, CTRL+F or CTRL+M, F12, F3, UP, DOWN work, but not CTRL+S