NicoHood/HID #204: Gamepad for Nintendo Switch

gdsports, 2020-01-23:
https://github.com/gdsports/HID/tree/horipad/examples/NSGamepad

Inspired by various NS gamepad projects using LUFA and AVR boards. SAMD has more capability such as capactive touch. Not much testing so far but buttons work.

gdsports, 2020-10-06:
This has a new home at https://github.com/gdsports/NSGadget_HID

sparten9999, 2021-04-26:
i gave the above link a try and just get "not declared in this scope" errors in all examples
nsgadget gives 'NSGamepad' was not declared in this scope
nsgamepad gives 'NSGAMEPAD_DPAD_CENTERED' was not declared in this scope
nsgamepad demo gives nsgadget gives 'NSGamepad' was not declared in this scope
nskeyboard gives 'NSGAMEPAD_DPAD_CENTERED' was not declared in this scope

closed 2021-04-27 by NicoHood