NicoHood/HID #205: No more than two SingleGamepads possible

MickGyver, 2020-02-14:
I can't use Gamepad3 and 4 without breaking the serial device on the Leonardo (I have to burn the bootloader again). I can enable up to two Gamepads, but any more and serial is broken and I have to burn the bootloader again. I'm using Arduino IDE version 1.8.9.

MickGyver, 2020-02-14:
I tested this library with HoodLoader2 also but it didn't help, same problem. Also tried version 1.8.12 and 1.6.7 of the Arduino IDE to no avail.

jroth, 2020-07-18:
@MickGyver 

I was digging through the wiki looking into this and found this note.

You can declare HID devices in two ways: Each device gets it own report or you smash them all together. Last open we used till 2.3. With 2.4 both options are supported and they have their pro and cons. All basic features should be possible with a multireport. You can have a look at the specific API documentation which report type is supported with what name (example: Keyboard vs BootKeyboard).

The difference you should care about is the endpoint limitation of the microcontrollers. The 32u4 has 6 USB endpoints and the 8/16/32u2 series only 4 (with less USB_EP_SIZE on top of that: 64 vs 16 bytes). This means that you can only add a limited number of devices. The CDC Serial uses 3 endpoints. **This means you can add up to 3 devices for the 32u4 and 1 for the 8/16/32u2. If you add more, some will be ignored.**

NicoHood, 2020-08-02:
Correct!

closed 2020-08-02 by NicoHood