NicoHood/HID #209: Can't compile, Multiple libraries were found for "HID-Project.h"

shadowbgd, 2020-02-26:
I get this error:
`
Arduino: 1.8.12 (Windows 10), Board: "SparkFun Pro Micro, ATmega32U4 (5V, 16 MHz)"

avr-g++: error: CreateProcess: No such file or directory

Multiple libraries were found for "HID-Project.h"
 Used: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\HID
 Not used: C:\Users\shadow\Documents\Arduino\libraries\HID
exit status 1
Error compiling for board SparkFun Pro Micro.
`

while trying to compile this sketch
'
#include <HID.h>

#include <HID-Project.h>
#include <HID-Settings.h>

#define ENCODER_DO_NOT_USE_INTERRUPTS
#include <Encoder.h>

// Change these two numbers to the pins connected to your encoder.
//   Best Performance: both pins have interrupt capability
//   Good Performance: only the first pin has interrupt capability
//   Low Performance:  neither pin has interrupt capability
Encoder myEnc(5, 6);
//   avoid using pins with LEDs attached

void setup() {
  Consumer.begin();
//  Serial.begin(9600);
//  Serial.println("Basic Encoder Test:");
}

long oldpos  = 0;

void loop() {
  long newpos = myEnc.read();
  if (newpos != oldpos) {
    if(newpos > oldpos) {
      Consumer.write(MEDIA_VOL_UP);
    } else {
      Consumer.write(MEDIA_VOL_DOWN);
    }
    }
    oldpos = newpos;
    
//    Serial.println(newPosition);
  }
}
'

shadowbgd, 2020-02-26:
Full error log

[arduino_error.txt](https://github.com/NicoHood/HID/files/4258186/arduino_error.txt)

shadowbgd, 2020-03-04:
Solved. For some reason, it didn't work by manual install in Documents\Arduino\libraries\ but with install from Arduino IDE works fine.

closed 2020-03-04 by shadowbgd

virtualimpure, 2021-06-05:
> 
> 
> Solved. For some reason, it didn't work by manual install in Documents\Arduino\libraries\ but with install from Arduino IDE works fine.

Great! This worked for me too, manual install failed, via lib install, it worked! Thank you!