NicoHood/HID #210: NKRO is no longer work in SAMD21

MaveplaHC, 2020-03-02:
NKRO functions are work in AVR but do not have the right function in the SAMD21 environment. I have tried the example but it doesn't work which can't output anything.

The official of Arduino has posted an update about Jan, 2019 and fixed the issues of their keyboard library. Maybe that can do some help.

gdsports, 2020-03-02:
This works for me.

```
diff --git a/src/HID-APIs/NKROKeyboardAPI.h b/src/HID-APIs/NKROKeyboardAPI.h
index 6ff0c1a..f027d26 100644
--- a/src/HID-APIs/NKROKeyboardAPI.h
+++ b/src/HID-APIs/NKROKeyboardAPI.h
@@ -30,12 +30,14 @@ THE SOFTWARE.
 // +1 reportID, +1 modifier, +1 custom key
 #define NKRO_KEY_COUNT (8*13)
 
-typedef union{
+#define ATTRIBUTE_PACKED  __attribute__((packed, aligned(1)))
+
+typedef union ATTRIBUTE_PACKED {
        // Modifier + keymap + 1 custom key
        uint8_t whole8[0];
        uint16_t whole16[0];
        uint32_t whole32[0];
-       struct{
+       struct ATTRIBUTE_PACKED {
                uint8_t modifiers;
                uint8_t keys[NKRO_KEY_COUNT / 8];
                uint8_t key;

```

NicoHood, 2020-08-02:
https://github.com/NicoHood/HID/pull/211

closed 2020-08-09 by NicoHood