NicoHood/HID #213: add haptic feedback for surface dial

Eddddddddy, 2020-03-24:
i saw the setting in the windows 10 has a swtich. The swtich can contral the haptic feedback for surface dial. so how can i enable the feedback for my arduino dial.

i need some help

ubergeekseven, 2020-09-30:
I would also like to understand this. I have tried finding any uses in the wild and cannot find anything myself. It seems as though it is available somehow though. Not sure if the haptics are generated on the device based off of what you would code them for or if windows sends back something to give specific haptic feedback. i could definitely add local haptic code myself but, if windows can do it, then I would like to know how that is done.

The only thing I could find for any haptic feedback with a dial is from a project that does not use this library and has several other things within it that I would rather use this library for customizing. His does have haptics, local to the device though. So that could be added to the sketch you want to use and then done when whatever it is you want feed back for is triggered. I assume that any button push for selecting is one haptic output and then turning the dial may be another. If you were to increase volume by each pulse generated, then send a vibration for each pulse.

Freds4hb, 2020-09-30:
I'm working on using my matrix switches to activate LED's when momentary lights are switched. This could likely be done for your haptics as well with an diode protection and frankly via an led  (diode) you could actuate a haptic (motor or solenoid) quickly I think (without it being part of the firmware). It's cheating the software but can be done.

ubergeekseven, 2020-09-30:
That's not a bad idea at all. Thanks for the suggestion.
I haven't even tried this myself yet but, I can always run other code in the sketch to use other libraries for it. I think he was wanting whatever capability the actual Microsoft device has. I assume the switch in windows turns the function on and off from the computer and doesn't send the haptic data to it. With an encoder, I could have every pulse send the key and trigger the haptic motor.

Freds4hb, 2020-09-30:
You're welcome. I've been trying to sort this for a sim racing project I'm on, but I needed lights to come on when the buttons are pressed. It falls into the same category and I'd been wondering about that as well for a similar project (different control methodology). This is one of those "do it in hardware" times.

Eddddddddy, 2020-10-03:
i have no idea with my issue. 
finally, i type some code in my arduino sketch to direct drive taptic engine. it is not bad. just can’t turn off that function in windows setting.
`if (enc_action > 0) {
        //rotate_r
        digitalWrite(TAPTIC, HIGH);
        delay(Taptic_r);
        digitalWrite(TAPTIC, LOW);
  }
  else if (enc_action < 0) {
        //rotate_l
        digitalWrite(TAPTIC, HIGH);
        delay(Taptic_r);
        digitalWrite(TAPTIC, LOW);
  }`

ubergeekseven, 2020-10-05:
I am not sure about this but, I do not know if the library supports it talking back to the microcontroller. I may be wrong about that though. After playing around with the setup over the weekend, I am not sure it works as well as the actual dial input either.
The click is an issue for me, unless I need to physically debounce the button or something. When using in photoshop, the first click brings up the menu and then when you go to select the function in photoshop, like the brush size, sometimes it selects and sometimes it brings up the dial menu again. I need to find a way to get the response every time or it becomes something that is more in the way than helpful. Maybe a capacitor would help. 

You could add a button or switch that enables the haptic feedback. At least you could get a work around in that way.