NicoHood/HID #218: Keyboard.h not working.

nikidziuba, 2020-05-06:
Idk if it's problem with HID or library but it doesn't work (UNO).
Code (i'm usin ir):
#include <Keyboard.h>
#include <IRremote.h>

const int RECV_PIN = 4;
IRrecv irrecv(RECV_PIN);
decode_results results;
unsigned long key_value = 0;

void setup(){
  Serial.begin(9600);
  irrecv.enableIRIn();
  irrecv.blink13(true);
}

void loop(){
  if (irrecv.decode(&results)){
 
        if (results.value == 0XFFFFFFFF)
          results.value = key_value;

        switch(results.value){
          case 0xFFA25D:
          Serial.println("CH-");
          delay(500);
          break;
          case 0xFF629D:
          Serial.println("CH");
          delay(500);
          break;
          case 0xFFE21D:
          Serial.println("CH+");
          delay(500);
          break;
          case 0xFF22DD:
          Keyboard.press(KEY_LEFT_FN);
          Keyboard.press(KEY_F10);
          delay(200);
          Keyboard.releaseAll(); 
          break;
          case 0xFF02FD:
          Serial.println(">>|");
          delay(500);
          break ;  
          case 0xFFC23D:
          Serial.println(">|");
          delay(500);
          break ;               
          case 0xFFE01F:
          Serial.println("-");
          delay(500);
          break ;  
          case 0xFFA857:
          Serial.println("+");
          delay(500);
          break ;  
          case 0xFF906F:
          Serial.println("EQ");
          delay(500);
          break ;  
          case 0xFF6897:
          Serial.println("0");
          delay(500);
          break ;  
          case 0xFF9867:
          Serial.println("100+");
          delay(500);
          break ;
          case 0xFFB04F:
          Serial.println("200+");
          delay(500);
          break ;
          case 0xFF30CF:
          Serial.println("1");
          delay(500);
          break ;
          case 0xFF18E7:
          Serial.println("2");
          delay(500);
          break ;
          case 0xFF7A85:
          Serial.println("3");
          delay(500);
          break ;
          case 0xFF10EF:
          Serial.println("4");
          delay(500);
          break ;
          case 0xFF38C7:
          Serial.println("5");
          delay(500);
          break ;
          case 0xFF5AA5:
          Serial.println("6");
          delay(500);
          break ;
          case 0xFF42BD:
          Serial.println("7");
          delay(500);
          break ;
          case 0xFF4AB5:
          Serial.println("8");
          delay(500);
          break ;
          case 0xFF52AD:
          Serial.println("9");
          delay(500);
          break ;      
        }
        key_value = results.value;
        irrecv.resume(); 
  }
}
Errors:

sketch_may06a:37:26: error: 'KEY_F10' was not declared in this scope

           Keyboard.press(KEY_F10);

                          ^~~~~~~

Znaleziono wiele bibliotek w "Keyboard.h"
Wykorzystane: C:\Users\niki\Documents\Arduino\libraries\Keyboard
Niewykorzystane: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\libraries\Keyboard
exit status 1
Nie znalezione 'Keyboard'. Czy twój szkic zawiera linię '#include <Keyboard.h>' ?
^
||
it says couldnt find keyboard. Does your sketch include line #include <Keyboard.h> ?

Sunuba, 2020-05-12:
Did you try Keyboard.press(0xCB);

nikidziuba, 2020-05-12:
Thanks, it works!

closed 2020-08-02 by NicoHood