NicoHood/HID #219: IDE version check does not work with Arduino-Makefile

eharvester, 2020-05-18:
./local-libs/NicoHoodHID/src/HID-Project.h:31:2: error: #error HID Project requires Arduino IDE 1.6.7 or greater. Please update your IDE.
 #error HID Project requires Arduino IDE 1.6.7 or greater. Please update your IDE.

I have 1.8.12, Arduino-Makefile extracts "1812" 
HID-Project.h expects equal or greater than "10607"

I'm not sure which one is the correct integer interpretation for the IDE version number.

matthijskooijman, 2020-05-18:
This is a bug in Arduino-Makefile I think. Since Arduino 1.6.0, the format has changed from 1 digit  to 2 digits per component, so 1.8.12 should be `-DARDUINO=10812`. See https://github.com/arduino/arduino-cli/blob/master/docs/platform-specification.md#global-predefined-properties for the official reference (`runtime.ide.version`).

eharvester, 2020-05-19:
Thanks for clarification.

closed 2020-05-19 by eharvester