NicoHood/HID #222: Is this project active and maintained?

madhephaestus, 2020-06-12:
There are open PR's with a patch for making the samd21 work #211 , and critical bugs in the RawHID for samd21 that have open issues #156 and no progress. 

@acamilo and I would like to use this stack as part of an education environment, but would need to maintain it in order to do so. If this project is going to be revived by its owners, then we will happily use the PR's for fixing the code, and allow the maintainers to manage release. 

If, however, this was a side project back in 2015 and is now abandoned, (no judgment man, iv done it), then we would likely fork this project, rename it and re-publish through Arduino Library manager and maintain our fork moving forward.

NicoHood, 2020-08-02:
Hey @madhephaestus sorry for the late response. As you already noticed I am not that much active anymore :-((( However I am still interested in keeping the project alive and avoid wild forking. If you wish to help out with the maintenance of the project, I would really appreciate it.

So how would we do it? I would like to still have a look at ongoing PRs, but I aint got the resources to actually test the code. If you can help out with reviews, that would be really nice. I also have no sam device, so I am unable to test anyways.

If you need a faster response, always email me ;-)

SukkoPera, 2020-08-28:
This is a very useful library, we absolutely need to keep it maintained and improve it.

I am not at all into low-level USB stuff, but I will help with a few things, mostly the Keyboard and Gamepad stuff.

I think we also need someone to improve the current documentation, as it's outdated and some parts are just missing.

@NicoHood: Can you come up with a few more points you think should be improved? Maybe you can create a few "help needed" issues and tag them #Hacktoberfest, we might manage to bring some attention to this :).

NicoHood, 2020-08-28:
I am happy to merge any incoming PR, if it is tested and the code quality is sufficient. I am unsure if "help needed" tags would... help. If you have any changes or improvements, just open a PR.

SukkoPera, 2020-08-28:
I definitely will.

I think at least a "help needed" for the docs could be opened.

closed 2022-02-23 by NicoHood