NicoHood/HID #229: Update documentation

NicoHood, 2020-08-28:
If anyone is willing to write some more docs, please let me know...

marcdraco, 2020-08-31:
Hi Nico,

I got here because I needed some docs... (groan) but I'm a writer by profession, though not necessarily a great one it should be said.

I'm hoping to use some of your library with a project to convert old "pot based" analog sticks to USB for cheapskates like me - so I have a need for some of the functions. Hence, assuming I can get the basics working, I'll document and leave examples for what I can.

Obviously this is on the condition that I can make it work which isn't a given as Covid-19's lockdowns have sent me do-lally-tap.

Fingers crossed.

mrpommegit, 2020-09-08:
I will be more thna happy to help on documentation for the gamepad (youtube or wiki) if I found out on to make it work. I'm able to get it detected as hid but still not able to use the pin from the IO MCU to make it detected as keypress or jostick move (I plan to use linear 10k  poentiometer on this one)