NicoHood/HID #235: Errors with Keyboard and ASCII

WyattRichards1709, 2020-10-24:
When I compile anything that includes "HID_Project.h", it produces the following error: https://imgur.com/a/Nz7lGvL

The last time I used this library was back in June and I see that some ASCII stuff was changed and that might be the issue. Anyone else having this issue?

NicoHood, 2020-11-05:
Did this change cause the issue? https://github.com/NicoHood/HID/pull/176/files

leseb83, 2021-04-07:
On my end when doing:

#define HID_CUSTOM_LAYOUT
#define LAYOUT_FRENCH

#include "HID-project.h"
...

I am getting these errors:
hid-project\src\keyboardlayouts\improvedkeylayouts.h:56:18: note: #pragma message: Using custom layout for keyboard modules
  #pragma message "Using custom layout for keyboard modules"

arduino\libraries\hid-project\src\hid-apis\KeyboardAPI.hpp:151:17: error: '_asciimap' was not declared in this scope
  if(k >= sizeof(_asciimap)/sizeof(_asciimap[0])){

Thanks for your help.

NicoHood, 2021-04-07:
@SukkoPera Do you have any idea about that issue?

@leseb83 Which library version are you using? Please update

SukkoPera, 2021-04-07:
@NicoHood I think this is the way we handle them :).

@leseb83 Are sure you are using the latest git master tree? Looks like you might be using a version where some files have been updated and others haven't.

leseb83, 2021-04-08:
[image]

leseb83, 2021-04-08:
OK, then I copied manually the git repo, and it compiles properly now.
Anyway I still have some issues with this simple script as it doesn't launch the calculator program as expected but simply write **calc.exe(** in the current editor... :

#define HID_CUSTOM_LAYOUT
#define LAYOUT_FRENCH

#include <HID-Project.h>
#include <HID-Settings.h>

// Utility function
void typeKey(int key){
  Keyboard.press(key);
  delay(50);
  Keyboard.release(key);
}

void setup()
{
  // Start Keyboard and Mouse
  AbsoluteMouse.begin();
  Keyboard.begin();

  // Start Payload
  Keyboard.press(KEY_LEFT_GUI);
  Keyboard.press(114);
  Keyboard.releaseAll();

  delay(500);

  Keyboard.print("calc.exe");

  typeKey(KEY_RETURN);

  // End Payload

  // Stop Keyboard and Mouse
  Keyboard.end();
  AbsoluteMouse.end();
}

// Unused
void loop() {}

NicoHood, 2021-04-08:
I can reproduce the issue, but German and Italian compile fine. WTF why?

SukkoPera, 2021-04-08:
Quoting [the docs](https://github.com/NicoHood/HID/wiki/Keyboard-API):

> The original key definitions still work, just make sure you use the name, not the number. Keyboard.write(0xB0); will not work, use Keyboard.write(KEY_RETURN); instead.

So your:

    Keyboard.press(114);

should become:

    Keyboard.press('R');

You might also have to introduce some small delays between pressed and releases.

Regarding the issue with 2.7.0, I think that one only has the US and German layouts, there is no official version with all the keymaps currently available in the git master tree yet.

NicoHood, 2021-04-08:
I've created a new release, I totally missed creating a new release.

closed 2021-04-08 by NicoHood

leseb83, 2021-04-08:
Ok, fine for the 'r'.
But the typeKey() doesn't work with KEY_RETURN it produces a parenthesis (at least in french).
I was able to make it work by typing this specifically :
Keyboard.press(KEY_RETURN);
delay(50);
Keyboard.release(KEY_RETURN);

Thanks for fast replies !!

NicoHood, 2021-04-08:
`Keyboard.println("calc.exe");` is your solution

SukkoPera, 2021-04-08:
Alternatively, it might work if you change the type of your `key` function parameter to `KeyboardKeycode`.

leseb83, 2021-04-11:
Thanks a lot, all of above is OK !