NicoHood/HID #236: Nano33 IoT support possible?

tikonen, 2020-11-19:
Hi,
The board as native USB support but it appears that the HID library or examples don't work on this board. What happens is that I get a composite device with two COM ports, latter has HID descriptor data segment and Device Manager shows error for it.

I also tried with Arduino built-in Keyboard (https://www.arduino.cc/reference/en/language/functions/usb/keyboard/) and it fails the same way. 

Thank you!

tikonen, 2020-11-19:
Probably related issue: https://github.com/arduino/ArduinoCore-samd/issues/423

jbkuma, 2021-03-28:
Nano 33 IoT uses an ESP32 as a UART.

bsperryn, 2021-03-31:
Not sure whether you're still having a problem with this, but I have the gamepad running on my Nano IOT with no problem that I can see.

tikonen, 2021-04-01:
This has been solved. I needed to remove the driver that Arduino board had installed for the Nano 33 IoT.
Details here https://github.com/arduino-libraries/Keyboard/issues/41

closed 2021-04-01 by tikonen