NicoHood/HID #242: Not compiling with Arduino SAMD boards version 1.8.10

wanthalf, 2020-12-21:
The library stopped compiling with Arduino SAMD Boards after upgrade to version 1.8.10 in the Boards Manager. The error is:

`Arduino/libraries/HID-Project/src/HID-Settings.h:107:10: fatal error: USB/PluggableUSB.h: No such file or directory
 #include "USB/PluggableUSB.h"`

It still compiles well with all versions up to 1.8.9., when downgraded. I am not sure if it is more a problem of the Arduino SAMD Boards libraries or this library.

(Tested in Arduino IDE 1.8.13 on macOS 11.1, compilation for Arduino Zero (Native USB Port)).

arunmjayan, 2021-01-20:
I am also facing the same problem. Did anyone overcome this issue, if yes, can you please share how?

NicoHood, 2021-01-20:
There is a fix: https://github.com/NicoHood/HID/pull/244

If anyone is willing to check avr support, I am happy to merge this.

closed 2021-03-09 by NicoHood