NicoHood/HID #243: Failing to compile for Mega

LoungeFlyZ, 2021-01-02:
Selected HoodLoader2 Mega 2560 in IDE 1.8.13 and am getting this when compiling...

` #error HID Project can only be used with an USB MCU.`

Am I doing something wrong? or is something not working quite right?  I have HoodLoader2 loaded on my mega.

LoungeFlyZ, 2021-01-02:
Perhaps my understanding of this project for devices like mega is incorrect?  Do you have to write the bridge yourself on the 16u2? If so is there an example that shows proxying HID? (i've seen the basic serial bridge example).

NicoHood, 2021-01-02:
Ufff. Sometimes I am reading some old documentation and my wording is awful XD
I assume that I was referring to the IDE integration, the Hoodloader2 Board files. Those are installed on your end.

What you need to do now is to write two programs:
* One for the 16u2
* One for the 2560

You have to do the communication yourself, there is no better example than the simple bridge example. I always wanted to do something better, but nothing was really relyable that I could release it to the public. I guess that is not the answer you'd wish to hear, but you'd need to do some coding on your own...

NicoHood, 2021-01-02:
I will close this, you can still reply here or send me an email.

closed 2021-01-02 by NicoHood

LoungeFlyZ, 2021-01-02:
Thanks for the clarification @NicoHood!  That makes sense now.  

Did Hoodloader1 do all that out of the box? from some of the youtube vids i saw it looks like they use Flip to drop a hex and that it all thats needed.

NicoHood, 2021-01-03:
Well, yes. Hoodloader1 was a firmware, not a bootloader (I named it wrongly, as I did not understand the whole thing that clear at that time). I used some custom protocol, which I have developed myself. I was filtering the serial output and extracted special hid commands. That worked pretty good, but is not super reliable. I have not looked at that sine a **very** long time.