NicoHood/HID #248: Modifier Keys in Matrix Code

Oreo456, 2021-01-26:
I'm working on building a keyboard using my arduino pro micro using this library. There is one major problem though. When I use the modifier keys, such as shift, it works fine if I use Keyboard.press(KEY_LEFT_SHIFT), however when I call shift from an array like Keyboard.press(keys[][]) it doesn't work. Space has a similar problem, it's fine when I do Keyboard.press(KEY_SPACE) but when I call it from an array it outputs , for some reason. All the other keys work fine with the array, it's just specifically the keys I can't call by putting parentheses around the character. I tried going back to arduino's keyboard.h library, and all the modifier keys work with that fine, so it's something about this library
Sorry, I'm not great at explaining things so sorry if this doesn't make sense.
Here's the code:
Thanks

`#include <HID-Project.h>
#include <HID-Settings.h>

#include <Adafruit_MCP23017.h>

Adafruit_MCP23017 mcp;
byte inputs[] = {4,5,6,7,8,9};
const int inCount = sizeof(inputs)/sizeof(inputs[0]);
byte outputs[] = {0,1,2,3,4,5,6,7,8,9,10,11,12,13,14};
const int outCount = sizeof(outputs)/sizeof(outputs[0]);

char keys[2][2] = {
  {KEY_LEFT_SHIFT,'j'},
  {'k','w'}
  };

bool keysDown[2][2] = {
  {false, false},
  {false, false}
};

void setup() {
  // put your setup code here, to run once:
  
  mcp.begin();
  for(int i=0; i<outCount; i++){    //declaring all the outputs and setting them high
    mcp.pinMode(outputs[i],OUTPUT);
    mcp.digitalWrite(outputs[i],LOW);
  }
  for(int i=0; i<inCount; i++){     //declaring all the inputs and activating the internal pullup resistor
    pinMode(inputs[i],INPUT_PULLUP);
  }
  Serial.begin(9600);
  Keyboard.begin();
}

  
void loop() {
  // put your main code here, to run repeatedly:
  keyCheck();
  

}

void keyCheck()
{
  for (int i=0; i<2; i++){
    mcp.digitalWrite(outputs[i],LOW);
    for (int j=0; j<2; j++)
    {
      if(digitalRead(inputs[j]) == LOW && keysDown[i][j] == false)
      {      
        Serial.print("Row: ");
        Serial.print(i);
        Serial.println();
        Serial.print("Col: ");
        Serial.print(j);
        Serial.println();
        Serial.print(keys[i][j]);
        Serial.println();
        if(keys[i][j] == KEY_LEFT_SHIFT)
        {
          Serial.print("Shift Key");
          Serial.println();
        }
        Keyboard.press(keys[i][j]);
        Serial.println();
        keysDown[i][j] = true;
        Serial.print("KeysDown set to true");
        Serial.println();
      }else if(digitalRead(inputs[j]) == HIGH && keysDown[i][j] == true)
      {
        Serial.print("keysdown set to false");
        Serial.println();
        Keyboard.release(keys[i][j]);
        keysDown[i][j] = false;
      }
      delay(1);
    }
    mcp.digitalWrite(outputs[i], HIGH);
  }
}`

gedankenexperimenter, 2021-01-27:
The problem here is a mismatch between ASCII and USB HID keycodes. `KEY_SPACE == 44 == ',' != ' '`; likewise, `'j' == 0x6A == KEY_F15 != KEY_J`. When you call `Serial.print()`, it interprets the argument as an ASCII character, but when you call `Keyboard.press()`, it's a USB HID keycode. So, your `keys` array should be defined like this:

```c++
char keys[2][2] = {
  {KEY_LEFT_SHIFT, KEY_J},
  {KEY_K, KEY_W}
};
```
…and when you call `Serial.println(keys[i][j])`, you shouldn't expect the output to contain the corresponding ASCII character.

Oreo456, 2021-01-27:
> The problem here is a mismatch between ASCII and USB HID keycodes. `KEY_SPACE == 44 == ',' != ' '`; likewise, `'j' == 0x6A == KEY_F15 != KEY_J`. When you call `Serial.print()`, it interprets the [...]

When I set up the array like that though, it doesn't do anything. The only way I can get a normal output is when I have the characters that have the '' around them, which I now know are the ASCII codes. The thing that confuses me the most is how come it works when I use Keyboard.press(KEY_LEFT_SHIFT), but not when I call KEY_LEFT_SHIFT from an array with Keyboard.press(keys[0][0]). Is there something I need to do with the array to make it compatible with the USB HID keycodes?
Thanks for your help

Riisko, 2021-02-06:
Experiencing the same problem, doesn't matter if my matrix is using byte or char, the KEY_F1 to KEY_F6 is translated to :;<=>? with shift and ctrl modifiers on and off.
I have used the original Keyboard library before and it worked correctly.

Oreo456, 2021-03-03:
I couldn't find a solution, but I found another way to do it, using a switch statement instead. Here's my code. Feel free to use it however you like Riiso

closed 2021-03-03 by Oreo456