NicoHood/HID #249: How to setup Game Controller with LEDs?

bert-laverman, 2021-01-30:
I would like to use an Arduino Micro as a GameController (I assume the `GamePad` does this) with LEDs. (which I saw in the `TeensyKeyboardAPI`) I found a C library to control HID devices from Windows which supports a `setLED(int, bool)`, but all repositories so far assume the reader knows what he is doing wrt USB communication, which I don't.

I'm not afraid of the programming, just worried about the required background knowledge on how HID devices communicate via USB, and how to extend/adapt this HID library to do what I want.

Goal is a controller that will be usable for flight simulation, but I have nice buttons with LEDs built-in, so I want to be able to tell the Arduino which LEDs to light up.

Cheers,
Bert

PS
Why does every library assume I want 32 buttons _and_ loads of axes _and_ Hat-switches? A constructor like:
```
GamePad::GamePad(int nButtons, int nAxes, int nHatswitches, int nLEDs);
```
would make a lot more sense...

NicoHood, 2021-01-30:
You need to communicate via Serial then, there is no easy way to send button states like that with this library or USB HID in general.

You cannot easily change the button count for a device, as (mostly windows) drivers will not handle changing devices properly. Thatswhy the maximum is used, to support all usecases. It does not hurt to have more buttons.

closed 2021-01-30 by NicoHood

bert-laverman, 2021-01-31:
The constructor is executed at startup, so it's not like I want to dynamically change the number of buttons. I just want a build to announce itself with exactly the right amount of components, and that works fine. When using another library I tried this, and when Windows accepted the device again it was shown with just three buttons and nothing else. No worries.

As for the LEDs; keyboards have them, so setting LEDs is definitely part of the HID profile.

Anyways, I liked your library better than the other one I tried, so I'll fork it and adjust. Been trying to read up on the USB specs and it's a real mess. Looks like no Software Engineers with modern client-server/networking experience have been involved with it, so the terminology used looks unique for USB. At least, I've never come across a networking protocol specified in terms of "pipes", with the "IN" and "OUT" pipe endpoint names assigned not through function or usage, but from the perspective of the "host computer". And no mention of "messages", but everything is a "report". Weird! Need to do a lot of translating before it starts making sense.

NicoHood, 2021-01-31:
The isse with button number comes from USB descriptors. To save RAM they are located in PROGMEM. And since you cannot dynamically change PROGMEM data, a fixed value of 32 buttons is used. You could work around this by using DEFINE statements, but this is not really supported in arduino from the ino file (unless that changed over time). If you need such a granular setting, you should rather use the LUFA usb core directly.

LEDs are supported by the OS, but mostly those 3 default leds (numlock, etc) with a **keyboard**. Now you could add leds to a controller, but I dont know if the OS will still recognize the device. I guess it will, but nothing happens with those LEDs. If you are using something like libusb you could of course control those LEDs. You could even use custom feature reports, but as you already noticed, there is very little documentation about that. This is nothing to put into a user friendly arduino library. I already tried that, but such complex usecases are not easy to abstract while still keeping the code size small.

Now if your flight simulator uses custom code anyways, I (still) recommend using the serial port, as this is already supported and working properly. Otherwise you'd be better using LUFA, but you need a lot of time to understand that stuff.

If you got something working, just let me know. ;-)