NicoHood/HID #257: Doesn't work with nRF528x boards such as Nano 33 and Nano 33 Sense

bsperryn, 2021-02-27:
When I try compile this for a Nano 33, I get the following error:

#error HID Project can only be used with an USB MCU

bsperryn, 2021-02-28:
I have found this library: https://github.com/arduino/ArduinoCore-mbed that seems to add the pluggableUSB functionality to these boards.

bsperryn, 2021-03-02:
I've installed the library I mentioned which has given me new core's to use that in theory allow pluggable USB to work. However I still get the following errors:

```In file included from /Users/briansperryn/Documents/Arduino/test_HID-project/test_HID-project.ino:1:0:
/Users/briansperryn/Documents/Arduino/libraries/HID-master/src/HID-Project.h:35:2: error: #error HID Project can only be used with an USB MCU.
 #error HID Project can only be used with an USB MCU.
  ^~~~~
In file included from /Users/briansperryn/Documents/Arduino/libraries/HID-master/src/HID-Project.h:39:0,
                 from /Users/briansperryn/Documents/Arduino/test_HID-project/test_HID-project.ino:1:
/Users/briansperryn/Documents/Arduino/libraries/HID-master/src/SingleReport/SingleAbsoluteMouse.h:28:10: fatal error: HID.h: No such file or directory
 #include "HID.h"
          ^~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Nano 33 BLE.```

NicoHood, 2021-03-04:
Try to remove that line and check if it works:
https://github.com/NicoHood/HID/blob/master/src/HID-Project.h#L35

itavero, 2021-05-05:
FYI: Just removing the check does **not** work:
```
In file included from /Users/arnom/Documents/Arduino/libraries/HID-Project/src/HID-Project.h:39:0,
                 from /Users/arnom/Documents/Arduino/libraries/HID-Project/examples/Consumer/Consumer.ino:14:
/Users/arnom/Documents/Arduino/libraries/HID-Project/src/SingleReport/SingleAbsoluteMouse.h:28:10: fatal error: HID.h: No such file or directory
 #include "HID.h"
          ^~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Nano 33 BLE.
```

NicoHood, 2021-05-05:
With which error message?

Maybe this PR can be related to porting also other MCUs that support pluggable usb:
https://github.com/NicoHood/HID/pull/244/files