NicoHood/HID #258: HID Project can only be used with an USB MCU.

ryankall82, 2021-03-03:
Trying to use a Teensy with Encoder.h and HID-Project for a volume knob.

In file included from C:\Users\ryank\Documents\Arduino\2knob\2knob.ino:9:0:
Multiple libraries were found for "Encoder.h"
C:\Users\ryank\Documents\Arduino\libraries\HID-Project\src/HID-Project.h:35:2: error: #error HID Project can only be used with an USB MCU.
 Used: C:\Users\ryank\Documents\Arduino\libraries\Encoder
 #error HID Project can only be used with an USB MCU.
 Not used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Encoder
  ^
In file included from C:\Users\ryank\Documents\Arduino\libraries\HID-Project\src/HID-Project.h:39:0,
                 from C:\Users\ryank\Documents\Arduino\2knob\2knob.ino:9:
C:\Users\ryank\Documents\Arduino\libraries\HID-Project\src/SingleReport/SingleAbsoluteMouse.h:28:17: fatal error: HID.h: No such file or directory
compilation terminated.
Error compiling for board Teensy 4.0.

#include <Encoder.h>

#include <HID-Project.h>
#include <HID-Settings.h>

// Change these pin numbers to the pins connected to your encoder.
//   Best Performance: both pins have interrupt capability
//   Good Performance: only the first pin has interrupt capability
//   Low Performance:  neither pin has interrupt capability
Encoder knobLeft(14, 6);
Encoder knobRight(18, 8);
//   avoid using pins with LEDs attached

void setup() {
  Serial.begin(9600);
  Serial.println("TwoKnobs Encoder Test:");
}

long positionLeft  = -999;
long positionRight = -999;

void loop() {
  long newLeft, newRight;
  newLeft = knobLeft.read();
  newRight = knobRight.read();
  if (newLeft != positionLeft || newRight != positionRight) {
    Serial.print("Left = ");
    Serial.print(newLeft);
    Serial.print(", Right = ");
    Serial.print(newRight);
    Serial.println();
    positionLeft = newLeft;
    positionRight = newRight;
  }
  // if a character is sent from the serial monitor,
  // reset both back to zero.
  if (Serial.available()) {
    Serial.read();
    Serial.println("Reset both knobs to zero");
    knobLeft.write(0);
    knobRight.write(0);
  }
}

NicoHood, 2021-03-04:
Does teensy support the arduino pluggable usb? If not, teensy is not supported

kkuez, 2021-04-10:
Got the same with an Arduino Nano. Should be supported shouldnt it?

NicoHood, 2021-04-10:
Of course it is NOT.

korapakisan, 2022-12-24:
hello,
Mada about arduino uno

closed 2022-12-25 by NicoHood