NicoHood/HID #264: Notes on keyboard support

SukkoPera, 2021-03-11:
I have prepared some notes about keyboard support that you might want to add to the relevant wiki page. I cannot send a PR for that, so I am including them here:

---

When working on a keyboard project, you first need to decide whether you want to work on a *character* or on a *key* basis.

For instance, if you want to make a big button that simulates the press of the <kbd>Y</kbd> key, you're better off working at the character level. This is because keyboards are not the same all over the world and while it might be natural for you to think of the <kbd>Y</kbd> key as sitting between your <kbd>T</kbd> and <kbd>U</kbd> keys, you might be surprised at knowing that this might not be the case for German people, as they have a <kbd>Z</kbd> in there. This means that the actual key corresponding to the character to want to type depends upon the particular [keyboard layout](https://en.wikipedia.org/wiki/Keyboard_layout) in use. Recent releases of the HID Library have support for the most common layouts, so that they will always press the key corresponding to the character you want to type. Just use:

```
Keyboard.write('Y');
```

For this to work, you will need to add the following at the top of your sketch:

```
#define HID_CUSTOM_LAYOUT
#define LAYOUT_GERMAN
```

Make sure to add it **before** the `#include <HID-Project.h>` directive. Of course you can customize the second line to choose your desired keylayout. You can have a glimpse at the available layouts [at the bottom of the *ImprovedKeylayouts.h* file](https://github.com/NicoHood/HID/blob/master/src/KeyboardLayouts/ImprovedKeylayouts.h#L659).

Keep in mind that **not all characters can be typed with all the keylayouts**. This is because the API does not support <kbd>AltGr</kbd> combined ASCII lookup tables nor characters requiring a *sequence* of keypresses to be entered (e.g.: accents on dead keys).

On the other hand, sometimes you want to press a particular key, regardless of what character it will print. This might be the case when working on a full keyboard, for instance: when the user presses the key between T and U, you will want to send the scancode assigned to that particular key by the USB keyboard specs, it will be the job of the host device to decide what character it corresponds to, according to the keymap it is configured for. In this case [the original Arduino key definitions](https://www.arduino.cc/en/Reference/KeyboardModifiers) still work, just make sure you use the name, not the number. `Keyboard.write(0xB0);` will not work, use `Keyboard.write(KEY_RETURN);` instead.

NicoHood, 2021-03-12:
This information was so much needed!!! Thanks!

A Picture would make a lot of sense. I think 3 keyboards confuse too much, can you maybe find an image that is also free to use (copyright)?

SukkoPera, 2021-03-13:
You can pick one [here](https://en.wikipedia.org/wiki/Keyboard_layout) or [here](https://en.wikipedia.org/wiki/QWERTY).

NicoHood, 2021-03-13:
Nice! I've added that!
https://github.com/NicoHood/HID/wiki/Keyboard-API

closed 2021-03-13 by NicoHood