NicoHood/HID #265: Notes on Gamepad support

SukkoPera, 2021-03-11:
Here's something for the Gamepad wiki page:

---

The Gamepad API supports 32 digital buttons, 4 16-bit axes (X, Y, RX, RY), 2 8-bit axes (Z, RZ) and 2 D-Pads/Hat switches.

As usual, there	are both single- and multi-report versions of the API.

The single-report version supports up to 4 Gamepads, named `Gamepad1`, `Gamepad2`, `Gamepad3` and `Gamepad4`. It will use 1 endpoint for each gamepad.

The multi-report version only suppors a single Gamepad, unsurprisingly named `Gamepad`.

All objects have the same API, which is straightforward:

## General
- `void begin(void)`: Initialize the gamepad. It is very important to call this function first on any Arduino type as it will send an initial clean report to the host.
- `void end(void)`: Terminate the gamepad instance. After this function is called, the gamepad can no longer be controlled.
- `void write(void)`: Send the actual report to the host. This function must be called to make any change effective.

## Digital Buttons
- `void press(uint8_t b)`: Press the `b`<sup>th</sup> button [0 .. 31]
- `void release(uint8_t b)`: Release the `b`<sup>th</sup> button [0 .. 31]
- `void releaseAll(void)`: Release all buttons
- `void buttons(uint32_t b)`: Set the state of all buttons at once

## Axes
- `void xAxis(int16_t a)`: Set the value of the X axis [-32768 .. +32767]
- `void yAxis(int16_t a)`: Set the value of the Y axis [-32768 .. +32767]
- `void zAxis(int8_t a)`: Set the value of the Z axis [-128 .. +127]
- `void rxAxis(int16_t a)`: Set the value of the RX axis [-32768 .. +32767]
- `void ryAxis(int16_t a)`: Set the value of the RY axis [-32768 .. +32767]
- `void rzAxis(int8_t a)`: Set the value of the RZ axis [-128 .. + 127]

## D-Pads/Hat Switches
- `void dPad1(int8_t d)`: Set the direction of the first D-Pad. `d` should be one of the following constants:
  - GAMEPAD_DPAD_CENTERED
  - GAMEPAD_DPAD_UP
  - GAMEPAD_DPAD_UP_RIGHT
  - GAMEPAD_DPAD_RIGHT
  - GAMEPAD_DPAD_DOWN_RIGHT
  - GAMEPAD_DPAD_DOWN
  - GAMEPAD_DPAD_DOWN_LEFT
  - GAMEPAD_DPAD_LEFT
  - GAMEPAD_DPAD_UP_LEFT
- `void dPad2(int8_t d)`: Set the direction of the second D-Pad, see above.

Please refer to [the Gamepad example](https://github.com/NicoHood/HID/blob/master/examples/Gamepad/Gamepad.ino) to see the API in action.

NicoHood, 2021-03-12:
Thanks a lot!
https://github.com/NicoHood/HID/wiki/Gamepad-API

closed 2021-03-12 by NicoHood

DRSDavidSoft, 2024-04-29:
I wish the wiki was open to edits by public people