NicoHood/HID #266: Reduce number of pre-defined Gamepad instances

SukkoPera, 2021-03-15:
https://github.com/NicoHood/HID/blob/7e3741426137a099f72ed692dec38c44e1bf0ec3/src/SingleReport/SingleGamepad.cpp#L149-L152

I think it would be better to reduce the number of pre-defined single-report Gamepad instances to 2.

Having 4 does not make much sense IMHO as even on the most capable 32U4 there are only 3 usable endpoints, so besides barring the possibility to use the Gamepad with one or two different devices, the 4 instances cannot even show up all together.

Having two - or even a single one - is not a limitation, since additional devices can be declared in user sketches, if more are needed.

NicoHood, 2021-03-15:
They cannot be declared in the user sketch, since they need a different id, or am I wrong?

SukkoPera, 2021-03-15:
Uhm... I cannot test right now, but as the declaration is just a simple `SingleGamepad_ GamepadX`, I don't see any problems with declaring them in user sketches.

Documenting how to disable the CDC serial is a good idea, but still it sounds odd to me to ship with something that requires fiddling with the Arduino source files to be usable. If someone is smart enough to do that, I'm sure they can also add/uncomment a few lines in the HID Library to enable as many `Gamepad`s. as they want (but still, I'm pretty convinced the latter can be done in the user sketch).

On the contrary, a beginner expects the library to work out of the box, so maybe they start using a Gamepad, then add a keyboard, which won't show up at all, to their surprise. This is because even if only Gamepad1.begin() is called, all the 3 available endpoints will instantly get used by Gamepads.

NicoHood, 2021-03-15:
> This is because even if only Gamepad1.begin() is called, all the 3 available endpoints will instantly get used by Gamepads.

I dont think this is true. If that is the case, we'd have problems since the last 3-4 years or so.

SukkoPera, 2021-03-15:
Exactly, I tested it and that's my experience.

That's what's causing the problems in #139, I suspect.

NicoHood, 2021-03-15:
Can this be solved by creating extra files with:

SingleGamepad1.cpp (and 2,3,4)
```cpp
#include "SingleGamepad.h"
SingleGamepad_ Gamepad1; // and 2,3,4
```

Then the compiler should only create those instances if they are referenced.

SukkoPera, 2021-03-20:
> Uhm... I cannot test right now, but as the declaration is just a simple `SingleGamepad_ GamepadX`, I don't see any problems with declaring them in user sketches.

I have tested this and it works fine. Still haven't had the time to test the rest though.

alijani1, 2021-03-24:
Please dont disable the extra end points.  I also think documenting how to disable cdc is best.

I also would love to see the multi-report endpoint working a multiple gamepads.  why limit the multi-report end point to just one gamepad?

NicoHood, 2021-03-24:
@alijani1 I think (but its been a long time) that it just did not work under windows in multireports.

SukkoPera, 2021-03-24:
@alijani1 Can you help testing what @NicoHood [proposed above](https://github.com/NicoHood/HID/issues/266#issuecomment-799494811)?

If that doesn't work, I am still convinced only 1/2 instances should be predefined. Then we would have docs saying "you can have up to 5 if you do the following: 1. disable cdc; 2. enable more in the lib", with the latter that could be made into just enabling a #define.

alijani1, 2021-03-25:
> @alijani1 I think (but its been a long time) that it just did not work under windows in multireports.

Perhaps it was windows 7 or 8.  I use other libraries with windows 10 and multi-report gamepad descriptors work fine on Windows 10.   In linux there is an issue as Linux only supports multiple HIDs on a single usb natively only if the quirk flag HID_QUIRK_MULTI_INPUT is set in its hid-ids list. [hid-quriks.c](http://github.com/torvalds/linux/blob/master/drivers/hid/hid-quirks.c) has all the VID/PID combinations that have the flag set.  if you dont use one of the preset ones,  you have to set the usb quirk flag manually in start up script.

alijani1, 2021-03-25:
> @alijani1 Can you help testing what @NicoHood [proposed above](https://github.com/NicoHood/HID/issues/266#issuecomment-799494811)?
> 

I will try, but I know in my case, I was able to use keyboard HID functions even after I do gamepad1.begin.   I have to double check as I may have disabled gamepad3 in that project.  Will let you know once I get to review and also try singlegamepad extra files.

SukkoPera, 2021-08-29:
> Can this be solved by creating extra files with:
> 
> SingleGamepad1.cpp (and 2,3,4)
> 
> ```c++
> #include "SingleGamepad.h"
> SingleGamepad_ Gamepad1; // and 2,3,4
> ```
> 
> Then the compiler should only create those instances if they are referenced.

I have finally tested this myself - not throughly though - and it seems to work fine.

SukkoPera, 2021-10-01:
It looks like the ability to [disable the CDC serial](https://github.com/arduino/ArduinoCore-avr/pull/383#event-5392487627) has finally been merged into the Arduino mainline. Let's hope for a release soon.

fermino, 2024-03-15:
@NicoHood would you be willing to accept a PR that allows disabling the automatic creation of instances via a compile-time flag? (I am in the current need of defining the instances myself and I'm patching it manually but if it might help others I can submit it!)
Thanks :)