NicoHood/HID #267: Keyboard.write(KEY_F13); misbehaves on windows?

el-germi, 2021-03-25:
the following code using the standard arduino keyboard library works as expected, sending an 'F13' keystroke, but using HID-Project.h causes 'h'.
On the 'ImprovedKeylayouts.h' line 172 it says that it's expected to not work but why tho, and is there a way to fix it whilst keeping the System.write(SYSTEM_SLEEP) and System.write(SYSTEM_WAKE_UP) functions?

using a 5v arduino pro micro(32U4) and no HID_CUSTOM_LAYOUT.
windows keyboard layout is Spanish qwerty

Am I doing something wrong?

NicoHood, 2021-03-29:
Can you post me a *minimal* example that demonstrates the issue? I'd expect that `Keyboard.write(KEY_F13)` should work.

el-germi, 2021-03-29:
I can't, it was entirely my fault
I have a simple de-bounce function `void type(int key) {
  Keyboard.write(key);
  delay(500);
}` but as it turns out, the keys are objects in an enum and not defines like in the original;
replacing it with `void type(KeyboardKeycode key) {
  Keyboard.write(key);
  delay(500);
}` fixed my issue.
I'm terribly sorry for wasting your time :c
I should've read the code before posting

closed 2021-03-29 by el-germi