NicoHood/HID #271: NicoHood HID bricked my device

x1unix, 2021-04-05:
Hello, I have an Arduino Pro Micro and I tried to flash a simple sketch and use NicoHood's HID library.

Device doesn't recognized in any system (Linux/Win) and all systems complain for USB device descriptor.

Some dmesg sample:
```
device descriptor read/64. error -71
```

Looks like the only solution here is try to reflash bootloader using other arduino, but mb it's nice to have to add
some kind of disclaimer about possible danger of using this lib.

x1unix, 2021-04-05:
**Update:** looks like builtin `Keyboard.h` from arduino did some mess when it was liked with HID library.

closed 2021-04-05 by x1unix

antoniomolram, 2021-05-16:
did you figure out how to fix this? I am having a similar issue

x1unix, 2021-05-16:
@antoniomolram I had both imported Arduino's standard keyboard library and NicoHood's libraries.

The solution was to remove Arduino Keyboard library from a project and reflash my project to Arduino board.

I've used SparkFun's Arduino Pro Micro which had some kind of recovery mode, which allows to delay sketch execution for a couple of seconds, which allowed me to reflash my sketch.