NicoHood/HID #277: Custom Symbol / non ascii character

MarcAhoy, 2021-04-12:
Hello, I want to make a keyboard with the symbols × and ÷, but I cannot find how I could do that. A shortcut for ÷ is alt + 0247, and I tried that in the code but it doesent work. Thanks, Marc
`Keyboard.press(KEY_LEFT_ALT);
    delay(500);
    Keyboard.write('0');
    delay(200);
    Keyboard.write('2');
    delay(200);
    Keyboard.write('4');
    delay(200);
    Keyboard.write('7');
    Keyboard.releaseAll();
    delay(600);`

NicoHood, 2021-04-13:
Hmmm, does it work when you manually enter that combination? Your code looks correct.

MarcAhoy, 2021-04-13:
Yes, it works perfectly when I do it

NicoHood, 2021-04-14:
Try to use strings instead of characters. 
```
Keyboard.write('0');
// new
Keyboard.write("0");
```

MarcAhoy, 2021-04-14:
Ok, got it working. Instead of using Keyboard.write("0") or Keyboard.write('0') I used the keypad name    
```
Keyboard.press(KEY_LEFT_ALT); 
    delay(100); 
    Keyboard.press(KEYPAD_0); 
    delay(100); 
    Keyboard.press(KEYPAD_2); 
    delay(100); 
    Keyboard.press(KEYPAD_4); 
    delay(100); 
    Keyboard.press(KEYPAD_7); 
    delay(700);
    Keyboard.releaseAll(); `
```

MarcAhoy, 2021-04-14:
You might want to include this in the wiki for people who want custom characters

NicoHood, 2021-04-14:
That is a good idea! I will reopen this issue.

Did you also try if the "normal" numbers on your **real** keyboard work? I am really confused that your first code did not work.

MarcAhoy, 2021-04-14:
Hmm, the top numbers don't work even if I do it manually,
so that is why the code wasn't working

NicoHood, 2021-04-15:
Does this work for you?
```
Keyboard.press(KEY_LEFT_ALT); 
delay(100); 
Keyboard.write(KEYPAD_0); 
delay(100); 
Keyboard.write(KEYPAD_2); 
delay(100); 
Keyboard.write(KEYPAD_4); 
delay(100); 
Keyboard.write(KEYPAD_7); 
delay(700);
Keyboard.releaseAll(); 
```

Or even without delay?
```
Keyboard.press(KEY_LEFT_ALT); 
Keyboard.write(KEYPAD_0); 
Keyboard.write(KEYPAD_2); 
Keyboard.write(KEYPAD_4); 
Keyboard.write(KEYPAD_7); 
Keyboard.releaseAll(); 
```

Where can I lookup those key combinations? Lets say I want to add a smiley like that: https://www.compart.com/de/unicode/U+1F642

MarcAhoy, 2021-04-15:
Yup, both of them work fine and they put ÷. You can find those codes by just looking them up for the character you want or use this table https://theasciicode.com.ar/