NicoHood/HID #287: (STM32 BluePill) Can't compile example: HID.h no such file

jarekdudzinski, 2021-05-10:
Arduino: 1.8.13 clean install + STM32duino (latest) + HID-Project library (latest)
Can't compile any of the examples, all the time I get:

.../HID-Project/src/HID-Project.h:39,
.../HID-Project/examples/Keyboard/ImprovedKeyboard/ImprovedKeyboard.ino:13:
.../HID-Project/src/SingleReport/SingleAbsoluteMouse.h:28:10: fatal error: HID.h: No such file or directory
   28 | #include "HID.h"

NicoHood, 2021-05-11:
I think it is the same issue as https://github.com/NicoHood/HID/issues/257

See that issue for a possible fix

jarekdudzinski, 2021-05-11:
Heh. I know the reason: I was trying to use library with official stm32duino, not that one mentioned in the wiki... : D
Any chances of making it to support official core? Can't find any library for USB mass storage and more advanced functionality...

NicoHood, 2021-05-12:
Open an issue at the upstream arduino library, reference this issue and request support. It should be quite easy to add

RakinRkz, 2023-05-30:
Is the issue fixed? because I am getting same error. Using all the latest version libraries.