NicoHood/HID #288: RP2040 / RPi Pico support?

saeugetier, 2021-05-23:
Is RP2040 support planned? Or does anybody work on that?

NicoHood, 2021-07-14:
Feel free to submit a PR.

CCurl, 2022-02-25:
I would love to add support for the Pico, but like Frodo in "Lord of the Rings", I do not know the way. Is there any resource that could provide some guidance?

Sqaaakoi, 2022-11-16:
was also wondering the same thing, I was looking for a library that supports Consumer Control so I could set the absolute volume level

mcuee, 2023-06-17:
For Raspberry Pi Pico, maybe you will have better luck with another library.
https://github.com/adafruit/Adafruit_TinyUSB_Arduino

I have no isssues running the example from the above repo using Pico.
https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples/HID/hid_generic_inout