NicoHood/HID #317: Horizontal mouse wheel

marekmasek, 2021-09-05:
Hi,
am I right that it's not possible to scroll mouse wheel horizontally with this library?
I wanted to add encoder or joystick to my arduino and use it for horizontal scrolling, but I see that there is only 1 parameter in `move` function for "scrolling the wheel".

In the past I had Apple Mighty Mouse with scroll ball instead of scroll wheel, so it was possible to scroll vertically and horizontally. 
https://en.wikipedia.org/wiki/Apple_Mighty_Mouse

nikita-tomilov, 2021-09-06:
Hi!
I second this question.

@marekmasek It is possible to scroll horizontally if you adjust the arduino's built-in mouse library.
I've made a separate repo with this library, and for me it works with this HID library. Feel free to test it: https://github.com/nikita-tomilov/HMouse

marekmasek, 2021-09-06:
@nikita-tomilov thanks, I tried your library, but it doesn't work for me. When I try to scroll horizontally, it's scrolling always to the right, doesn't matter if the passed value to the method is negative/positive. I also tried vertical scrolling with your library and it's scrolling up/down and at the same time to the right. 
I tested this simple code with Arduino Mouse library and the vertical scrolling was working fine there. So I think that the problem is not in my code:
```
#include <HMouse.h>
#include <ClickEncoder.h>
#include <TimerOne.h>

// Rotary encoder connections
#define ENCODER_CLK 4
#define ENCODER_DT 3
#define ENCODER_SW 2

ClickEncoder *encoder;
int16_t last, value;

// Capture rotary encoder pulses
void timerIsr()
{
  encoder->service();
}

void setup() {
  encoder = new ClickEncoder(ENCODER_DT, ENCODER_CLK, ENCODER_SW, 4);
  Timer1.initialize(1000);
  Timer1.attachInterrupt(timerIsr);

  HMouse.begin();
  last = -1;
}

void loop() {
  value += encoder->getValue();
  if (value != last) {
    uint16_t diff = abs(value - last);
    signed char wheel = (last < value) ? 4 : -4;
    for (uint8_t i = 0; i < diff; i++) {
      HMouse.move(0, 0, 0, wheel);
    }
    last = value;
  }
}
```

nikita-tomilov, 2021-09-07:
@marekmasek Hello, sorry, I've made a very stupid bug in the library. I fixed it, and now the code shall work as expected. Feel free to update the lib and test the code once again.

My working example (that did not work before the bugfix, I had the same scroll-right-only behaviour):
```
#include "HMouse.h"

void setup(){
  Serial.begin(115200);
  while (!Serial) {};
  Serial.println("start ok");
  HMouse.begin();

  for (int i = 0; i < 10; i++) {
    HMouse.move(0, 0, 0, -4);
    delay(500);
    HMouse.move(0, 0, 0, 4);
    delay(500);
  }
}

void loop() {}
```

NicoHood, 2021-09-12:
I am happy to integrate that, if you create a PR for that.

WaGi-Coding, 2022-12-26:
> @marekmasek Hello, sorry, I've made a very stupid bug in the library. I fixed it, and now the code shall work as expected. Feel free to update the lib and test the code once again.
>
> My working [...]

can you make a pull request to finally add it into HID-Project? Would really appreciate it <3