NicoHood/HID #346: State format 'SBIT' is not supported as floating-point format (Unity3D Game engine)

jonathan84clark, 2022-03-02:
It appears that the Unity game engine doesn't like something about the way this library sets up the Arduino as a Gamepad.
I am using an Arduino Pro-micro board. When I connect it to Unity (using the new input system version 1.0.2 it throws this error:
Could not create a device for 'Arduino LLC Arduino Leonardo (HID)' (exception: System.Exception: State format 'SBIT' is not supported as floating-point format

I was able to dig into the Unity code a little. It looks like one of the values the Unity Game engine is getting from the gamepad is a floating point number -1.525879E-05. It could be a joystick axis, (not sure), it doesn't seem like that value is correct. I'm wondering if the joystick axis isn't being set to zero properly?

If you have any way to solve this or any recommendations that would be great. If you need anything else from me please let me know. I'm also not sure if this is an issue with this library or with Unity. 

Steps to reproduce
1. Upload the Gamepad example sketch to an Arduino pro-micro or Arduino Leonardo
2. Validate that Windows recognizes the board as an HID device
3. Disconnect the Arduino board from the computer
4. Install Unity 2019.4 LTS
5. Create a new Unity3D project and open it
6. In Unity go to Window -> Package Manager
7. Select "Input System" version 1.0.2 
8. Click Install (bottom right corner of the Windows)
9. Now connect the Arduino pro-micro or Arduino Leonardo to the computer
10. The error can be seen in the console and on the bottom of the main window

If you need to debug the Input System package to see what Unity is seeing you will need to create a local copy of the Input System package in another location, then in the package manager click on the + icon and select "Add Package from Disk" then select the "package.json" file that is within the Input System folder.

The input system folder is located here /YourUsername/AppData/Local/Unity/cache/packages/packages.unity.com/com.unity.inputsystem@1.0.2

I know this is a big ask but it would be awesome if I could build my own game controllers to play my own games!

Thanks,

Jonathan L Clark

jonathan84clark, 2022-03-02:
Thanks for the help! I will take a look at this; see if I can adapt those changes.

mdevaev, 2022-04-03:
Positive-only values really might help. Keep us posted please.

jonathan84clark, 2022-04-12:
Thanks again for the help. Sorry for not responding in so long (I had a bunch of other stuff going on).

I finally got a chance to look at the code. Are you saying that I need to change int16_t xAxis and int16_t yAxis (and other axis) to uint16_t values? I also need to modify Gamepad.cpp PROGMEM values? 

The reason I ask is the code in merge request #306 is a little bit of a different format. I'm still trying to wrap my head around how this all works. I'm still trying to understand HID. 

Also I'm still trying to understand HID. I'm fine creating USB-serial devices for PC but I haven't found a lot of documentation that explains the data format of HID controllers. Do you have a reference document you used to build this driver?

Once again, thanks for all the help! I've always wanted to create my own gamepad and your driver is the only open-source software I've seen that actually does it.

NicoHood, 2022-04-13:
You need to do exactly what the PR did. Modify the PROGMEM value. The xAxis should be still int16_t, as this is (for me) the better api, as it has a zero value. But the PC Interface (HID) should be from zero to positive values, as the PC will read it better. so we must do a mapping. Maybe this will solve your problem.