NicoHood/HID #361: Arduino Zero Compatibility?

GamerSocke, 2022-05-24:
Hi, 
After flashing the following code to the Zero via the native USB port, it does not show up as a gamepad like the Uno with the Hoodloader2.
Is there anything I need to be aware of when using the Zero?

```
#include "HID-Project.h"
void setup() {
    analogReadResolution(12);
    Gamepad.begin();
}

void loop() {
    int data = analogRead(A0);
    Gamepad.rxAxis(data);
    Gamepad.write();
    delay(10);
}

NicoHood, 2022-05-24:
I don't know. I remember such a bug report, but I cannot find it.

GamerSocke, 2022-05-25:
How can I investigate the problem further, do you have any idea what I could start with?

NicoHood, 2022-05-27:
Does the bootkeyboard work? Does the official arduino keyboard library work?

hlcmOxO, 2022-06-22:
I have a same question too, looks like neither RawHID and Joystick works properly on SAMD21.
But NKROKeyboard and mouse works fine.

mcuee, 2023-06-17:
For SAMD21, maybe you will have better luck with another library.
https://github.com/adafruit/Adafruit_TinyUSB_Arduino

I have no isssues running the example here from hidapitester.
https://github.com/todbot/hidapitester/tree/master/test_hardware/hidtest_tinyusb