NicoHood/HID #362: Gamepad API button numbers are inconsistent with the documentation

Clinery1, 2022-06-06:
# Problem
In the wiki, the `Gamepad` `press` and `release` functions say buttons `0..31`, but the actual API accepts `1..32`.

# Solution 1
Change the documentation to say `[1 .. 32]` instead of `[0 .. 31]`.

# Solution 2
Change the `Gamepad` API to support the buttons starting from 0 instead of 1.

# How to recreate
This snippet of code will (according to [gamepad-tester.com](https://gamepad-tester.com)) report `button0` when either `button0` or `button1` are pressed.
Tested on an Arduino Leonardo and Linux 5.17.
```c++
#include "HID-Project.h"
void setup() {
    Serial.begin(115200);
    pinMode(2,INPUT);
    pinMode(3,INPUT);
    Gamepad.begin();
}
void loop() {
    Gamepad.releaseAll();
    if (digitalRead(2)==HIGH) {
        Serial.println("Button 0 is pressed");
        Gamepad.press(0);
    }
    if (digitalRead(3)==HIGH) {
        Serial.println("Button 1 is pressed");
        Gamepad.press(1);
    }
    Gamepad.write();
    delay(20);
}
```
If the `press(0)` is changed to a `press(1)` and the `press(1)` to a `press(2)`, then gamepad-tester will report button 0 and 1 being pressed.

NicoHood, 2022-06-07:
What solution would you prefer?

Clinery1, 2022-06-08:
I think that just the documentation update is the better option.

NicoHood, 2022-06-08:
This page, 2 occurences, right?
https://github.com/NicoHood/HID/wiki/Gamepad-API

Clinery1, 2022-06-08:
That is correct.

NicoHood, 2022-06-09:
Done. Close if it is okay for you. Thanks for reporting