NicoHood/HID #377: send wrong key for enter as "("

vincentgao-ca, 2022-09-21:
Keyboard.write( KEY_ENTER);
BootKeyboard.write( KEY_ENTER);

Arduino 1.8.19, HID 2.8.2

I tested many times, and saw all the issues, no way to solve this.
Before I changed to nicohood hid library, everything is ok, but after I changed to NicoHood HID, the host receives "(", instead of ENTER. For the other keys, it works well.
Also I used it use Surface UEFI keyboard, not working at all, the Surface UEFI not working but it works well when plugging a real keyboard.

Please give a solution for how to use ENTER key and how to support Surface UEFI. Thanks.

NicoHood, 2022-09-22:
Hi,
which keyboard layout do you use?

vincentgao-ca, 2022-09-22:
Hi NicoHood,
I use  US_ENGLISH, Sparkfun Pro micro, 32u4 nano. I am trying to fix this issue now. I used Keyboard.write(KEY_ENTER) it is correct, I can't use uint8_t as input, the host will get "(".
For UEFI, still not working, either Keyboard or BootKeyboard class functions.
Thank you for your reply so quickly.
vincent

NicoHood, 2022-09-22:
I have no idea what is going wrong

vincentgao-ca, 2022-09-22:
I have to translate all the virtual code into Arduino code, especially for control, alt, Fn keys.
About supporting the UEFI keyboard, do you have any idea for solving the issue? Thank you.

closed 2022-09-22 by vincentgao-ca

vincentgao-ca, 2022-09-22:
I finally made it, translate each virtual key to the defined keys ascii map. 
Thank you.