NicoHood/HID #390: KEY_H issue

sirm15, 2022-12-20:
I'm having that error when I tried to compile for an arduino Leoardo, using the spanish layout:

In file included from c:\users\zarzu\documents\arduino\libraries\hid-master\src\hid-apis\KeyboardAPI.h:29:0,
                 from c:\users\zarzu\documents\arduino\libraries\hid-master\src\hid-apis\defaultkeyboardapi.h:27,
                 from C:\Users\zarzu\Documents\Arduino\libraries\HID-master\src/SingleReport/BootKeyboard.h:30,
                 from C:\Users\zarzu\Documents\Arduino\libraries\HID-master\src/HID-Project.h:50,
                 from C:\Users\zarzu\Documents\Arduino\sketch_dec20a\sketch_dec20a.ino:4:
c:\users\zarzu\documents\arduino\libraries\hid-master\src\keyboardlayouts\improvedkeylayouts.h:73:25: error: expected identifier before '=' token
     KEY_H               = 11,
                         ^
c:\users\zarzu\documents\arduino\libraries\hid-master\src\keyboardlayouts\improvedkeylayouts.h:73:25: error: expected '}' before '=' token
c:\users\zarzu\documents\arduino\libraries\hid-master\src\keyboardlayouts\improvedkeylayouts.h:73:25: error: expected unqualified-id before '=' token
c:\users\zarzu\documents\arduino\libraries\hid-master\src\keyboardlayouts\improvedkeylayouts.h:521:1: error: expected declaration before '}' token
 };
 ^

exit status 1

Compilation error: exit status 1

NicoHood, 2022-12-21:
I honestly have no idea. I checked the source and it all seems fine. Did you try other languages that do work? If yes, we could compare them and better debug

sirm15, 2022-12-21:
I've checked with other languages and still giving me the same error ( note the first pragma error ): 

In file included from c:\users\zarzu\documents\arduino\libraries\hid-project\src\hid-apis\KeyboardAPI.h:29:0,
                 from c:\users\zarzu\documents\arduino\libraries\hid-project\src\hid-apis\defaultkeyboardapi.h:27,
                 from C:\Users\zarzu\Documents\Arduino\libraries\HID-Project\src/SingleReport/BootKeyboard.h:30,
                 from C:\Users\zarzu\Documents\Arduino\libraries\HID-Project\src/HID-Project.h:50,
                 from C:\Users\zarzu\Documents\Arduino\sketch_dec20a\sketch_dec20a.ino:4:
c:\users\zarzu\documents\arduino\libraries\hid-project\src\keyboardlayouts\improvedkeylayouts.h:54:21: note: #pragma message: Using default ASCII layout for keyboard modules
     #pragma message "Using default ASCII layout for keyboard modules"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\users\zarzu\documents\arduino\libraries\hid-project\src\keyboardlayouts\improvedkeylayouts.h:73:25: error: expected identifier before '=' token
     KEY_H               = 11,
                         ^
c:\users\zarzu\documents\arduino\libraries\hid-project\src\keyboardlayouts\improvedkeylayouts.h:73:25: error: expected '}' before '=' token
c:\users\zarzu\documents\arduino\libraries\hid-project\src\keyboardlayouts\improvedkeylayouts.h:73:25: error: expected unqualified-id before '=' token
c:\users\zarzu\documents\arduino\libraries\hid-project\src\keyboardlayouts\improvedkeylayouts.h:521:1: error: expected declaration before '}' token
 };
 ^

exit status 1

Compilation error: exit status 1

If you need more info about the code or something, let me know

NicoHood, 2022-12-22:
Maybe something has changed with a recent avr code or the new ide. Are you using the new ide 2.0?

sirm15, 2022-12-22:
yeah, I'm using it, but I tried with the legacy arduino editor and still giving the same problem. When I checked the code from the .h file, it seems to be perfect with no error, I don'y know why it keep failing

NicoHood, 2022-12-22:
Maybe another library already defined the KEY_H token?

sirm15, 2022-12-22:
I'm using this libraies: Keypad.h, Encoder.h, Bounce2.h and HID-Project.h
And when I delete de KEY_H definition, other part of your library claims that is missing that definition, so I don't really think that it's been defined twice

sirm15, 2022-12-22:
If you wanna try, I could pass you my code, its a simple macropad script

NicoHood, 2022-12-22:
Sorry, I aint got time to debug this. But I guess you got the answer now.

sirm15, 2022-12-22:
not yep, I'll keep triying, and for sure I'll let ypu kn ow if I could fix it
Thanks!

NicoHood, 2022-12-24:
Let me know if you found a solution. I will close this issue for now, but you can still comment

closed 2022-12-24 by NicoHood

LucMoriarty, 2024-03-04:
Hello,

Got the same problem, this was due to #include <Keyboard.h> in my script.

Hope it will help some people!